zephyrproject-rtos/zephyr #14603: pyocd can't support more board_runner_args

lgl88911, 2019-03-17:
**Describe the bug**
Today I updated the master-branch code. zephyr using more args for pyocd will fail. last Year(2018)haven't the problem。

**To Reproduce**
Steps to reproduce the behavior:
1. add more args in board.cmake
```
board_runner_args(pyocd "--target=nrf52")
board_runner_args(pyocd "--flashtool-opt=-f 8000000")
board_runner_args(pyocd "--flashtool-opt=-fp")
include(${ZEPHYR_BASE}/boards/common/pyocd.board.cmake)
```
2. do make flash, you can see:
```
Using runner: pyocd
['--dt-flash=y', '--target=nrf52', '--flashtool-opt=-f 8000000', '--flashtool-opt=-fp']
Traceback (most recent call last):
  File "/home/frank/work/project/zephyrproject/.west/west/src/west/main.py", line 625, in <module>
    main()
  File "/home/frank/work/project/zephyrproject/.west/west/src/west/main.py", line 576, in main
    args.handler(args, unknown)
  File "/home/frank/work/project/zephyrproject/.west/west/src/west/main.py", line 331, in ext_command_handler
    command.run(*west_parser.parse_known_args(argv))
  File "/home/frank/work/project/zephyrproject/.west/west/src/west/commands/command.py", line 85, in run
    self.do_run(args, unknown)
  File "/home/frank/work/project/zephyrproject/zephyr/scripts/west_commands/flash.py", line 32, in do_run
    'ZEPHYR_BOARD_FLASH_RUNNER')
  File "/home/frank/work/project/zephyrproject/zephyr/scripts/west_commands/run_common.py", line 227, in do_run_common
    'received unknown arguments', unknown)
TypeError: __init__() takes from 1 to 2 positional arguments but 5 were given

```
3. if remove more args, make flash work well,
```
board_runner_args(pyocd "--target=nrf52")
#board_runner_args(pyocd "--flashtool-opt=-f 8000000")
#board_runner_args(pyocd "--flashtool-opt=-fp")
include(${ZEPHYR_BASE}/boards/common/pyocd.board.cmake)
```

**Expected behavior**
Pyocd script of zephy support more args

MaureenHelm, 2019-03-18:
Hi @lgl88911 ,  I changed the name of the argument from `flashtool-opt` to `flash-opt` in commit 987313e102cb36e007e89cb7c9147b204694e3f2. This was to make the runner consistent with recent changes in pyocd itself (e.g., moving from `pyocd-flashtool` to `pyocd flash`). Sorry for the inconvenience.

lgl88911, 2019-03-23:
@MaureenHelm

closed 2019-03-23 by lgl88911