zephyrproject-rtos/zephyr #14604: BLE disconnection caused by channel map request or connection parameter update request

jeff-setec, 2019-03-18:
**Describe the bug**
A clear and concise description of what the bug is.
What have you tried to diagnose or workaround this issue?

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context about the problem here.

jeff-setec, 2019-03-18:
[image]

jeff-setec, 2019-03-18:
[20190307-link_lost_channel_req.zip](https://github.com/zephyrproject-rtos/zephyr/files/2976522/20190307-link_lost_channel_req.zip)

jhedberg, 2019-03-18:
Can you reproduce this with the latest master branch? Most people are busy stabilising it for the 1.14 release currently, so if this issue is applicable there it may get prioritised better.

cvinayak, 2019-03-18:
@jeff-setec This is a limitation due to overlapping connection events in slave role. As the individual masters have no knowledge of other connection in the slave, they use the default instant offset of 6 events. If 4 connection events in the slave overlap, each connection listens to its master on the 4th connection event, if the event that the slave is listening fails to synchronize with the master, it has 2 more (forced) consecutive connection interval to listen, and if the packet from master is received beyond 6 intervals, then the slave will disconnect with reason instant passed.
I will analyze the logs later today, but could you let me know the zephyr board you are using (as 50 cm is too low a range you are observing), the reason for the missed instant could be due to poor radio conditions, like noisy 2.4GHz environment.

jeff-setec, 2019-03-19:
@jhedberg and @cvinayak 
Thanks! 

The Zephyr is running on a BLE module Laird BL652-SA-01, which is mounted on the board we designed, and is working as a slave connecting simultaneously to 4 mobile devices, 1 iPhone 7 Plus, 1 iPhone XR, 1 LG-X240YK, and 1 android tablet. All mobile devices are running an app we designed.

The disconnection occurs more between the Zephyr BLE module and the LG phone, and it seems all because of the same disconnection reason as I raised here.

carlescufi, 2019-03-19:
Switching this to feature request because this is an inherent limitation on the Bluetooth spec. We will analyze a bit further but it looks like the timing is imposed by the centrals (mobile phones) over which we have no control.

carlescufi, 2019-03-19:
@jeff-setec can you add traces for all 4 connections? you would need to use 4 sniffers in parallel. As long as you have 4 Nordic DKs you should be able to do that.

joerchan, 2019-03-20:
@jeff-setec I have looked at the sniffer logs you gave and there seems to be a bug here.
Could you log print the variables (instant in the packet, and conn->event_counter) used in calculating that the instant has passed with a sniffer capture for that connection?

joerchan, 2019-03-20:
@jeff-setec Also if possible could you check if the issue is reproducible on v1.14, and use the v2 of the nrf sniffer?

carlescufi, 2019-03-20:
Setting to low due to the uncertainty about the root cause.

carlescufi, 2019-08-23:
@jeff-setec could you please let us know if this is still an issue on the current master?

carlescufi, 2019-08-27:
Closing due to inactivity from reporter

closed 2019-08-27 by carlescufi