zephyrproject-rtos/zephyr #14605: mimxrt1020_evk/mimxrt1050_evk/mimxrt1060_evk cpp_synchronization meets Hardware exception

hakehuang, 2019-03-18:
**Describe the bug**
Hardware exception

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=minxrt1060_rvk ..
3. make
4. See error

**Expected behavior**
application runs well

**Impact**
cpp support

**Screenshots or console output**
```
***** USAGE FAULT *****

  Attempt to execute undefined instruction

***** Hardware exception *****

Current thread ID = 0x8000169c

Faulting instruction address = 0x6000504c

Fatal fault in thread 0x8000169c! Aborting.

***** HARD FAULT *****

  Fault escalation (see below)

***** BUS FAULT *****

  Instruction bus error

***** Hardware exception *****

Current thread ID = 0x8000169c

Faulting instruction address = 0x61549f66

Fatal fault in ISR! Spinning...
```

**Environment (please complete the following information):**
 - OS: (Linux,)
 - Toolchain (Zephyr SDK)
 - Commit SHA or Version used: v1.14.0_rc2

hakehuang, 2019-03-18:
@MaureenHelm , @dleach02

dleach02, 2019-03-18:
Not able to reproduce.

hakehuang, 2019-03-20:
@rt1020 and rt1060 does not have such issue for this case. this is strange

MaureenHelm, 2019-03-25:
I'm not able to reproduce either

hakehuang, 2019-03-28:
can some one try my binary? I can always reproduce it on rt1060 boards

[evkmimxrt1060_cpp_synchronization_flexspi_nor_zephyr.zip](https://github.com/zephyrproject-rtos/zephyr/files/3016100/evkmimxrt1060_cpp_synchronization_flexspi_nor_zephyr.zip)

MaureenHelm, 2019-03-28:
> can some one try my binary? I can always reproduce it on rt1060 boards
> 
> [evkmimxrt1060_cpp_synchronization_flexspi_nor_zephyr.zip](https://github.com/zephyrproject-rtos/zephyr/files/3016100/evkmimxrt1060_cpp_synchronization_flexspi_nor_zephyr.zip)

I tried it and it works.
```
***** Booting Zephyr OS 1.14.0-rc2 *****
Create semaphore 0x8000009c
Create semaphore 0x80000088
main: Hello World!
coop_thread_entry: Hello World!
main: Hello World!
coop_thread_entry: Hello World!
main: Hello World!
coop_thread_entry: Hello World!
main: Hello World!
coop_thread_entry: Hello World!
```

hakehuang, 2019-03-29:
@MaureenHelm , thanks a lot, Ineed check my board for this issue.

closed 2019-03-29 by hakehuang

hakehuang, 2019-04-02:
@MaureenHelm I find it is my board qspi has some issues when program in sector data. if I continues program pages in a sector then there will be some issues with the data in qspi. now I use a slow program algorithm which can resolve this issue.

dleach02, 2019-04-02:
@hakehuang how are you controlling the program algorithm?

hakehuang, 2019-04-22:
@dleach02 , I write a qspi program application to write the qspi flash, which is based on the qspi_polling demo in sdk.