zephyrproject-rtos/zephyr #14606: mimxrt1050_evk tests/kernel/fp_sharing kernel.fp_sharing fails

hakehuang, 2019-03-18:
**Describe the bug**
float point sharing test fails

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=mimxrt1050_evk ..
3. make
4. See error

**Expected behavior**
test PASS

**Impact**
fp usage

**Screenshots or console output**
```
***** Booting Zephyr OS 1.14.0-rc2 *****

Floating point sharing tests started

===================================================================

FAIL - calculate_pi_low@89. Computed pi %.6f, reference pi %.6f

FAIL - load_store_low

===================================================================

===================================================================

PROJECT EXECUTION FAILED

Pi calculation OK after 50 (high) + 0 (low) tests (computed %.6f)

Load and store OK after 100 (high) + 28782 (low) tests

Pi calculation OK after 150 (high) + 0 (low) tests (computed %.6f)

Load and store OK after 200 (high) + 28782 (low) tests
```

**Environment (please complete the following information):**
 - OS: (Linux)
 - Toolchain (Zephyr SDK,)
 - Commit SHA or Version used: v1.14.0_rc2

hakehuang, 2019-03-18:
@MaureenHelm , rt1020/rt1060 are ok with this case

agansari, 2019-03-19:
@hakehuang i am able to reproduce this

MaureenHelm, 2019-03-20:
git bisect points to:

```
commit 525065dd8b9e57c53012572c5d7122643ede3970 (HEAD, refs/bisect/bad)
Author: Andrew Boie <andrew.p.boie@intel.com>
Date:   Thu Jan 31 15:31:18 2019 -0800

    tests: convert to use app shared memory
    
    CONFIG_APPLICATION_MEMORY was a stopgap feature that is
    being removed from the kernel. Convert tests and samples
    to use the application shared memory feature instead,
    in most cases using the domain set up by ztest.
    
    Signed-off-by: Andrew Boie <andrew.p.boie@intel.com>
```

hakehuang, 2019-03-21:
frdmk64f also has problem with this case. but for frdmk64f there is no any console output

agansari, 2019-03-21:
Went back to 525065dd8b9e57c53012572c5d7122643ede3970 commit but this issue still happens.
Went back to previous tag v1.14.0-rc1, still happens.
Went back to tag v.1.13.0 but sdk or toolchain doesn't match anymore.
This issue is older than expected i think and finding the source will not be easy.

Also K64F is slooooowwwww........ it prints, but ot takes a long time, i.MX family is alot faster.

ioannisg, 2019-03-21:
Very similar failure to what we see in #9656 , @agansari

agansari, 2019-03-21:
@ioannisg i'm still debugging this issue.
I've observed that that reading FPSCR register (FPU status/ctrl) in _calculate_pi_high_ task had all kinds of exceptions. Adding ___set_FPSCR(0);_ at the begining of _calculate_pi_high_ passes the test (reset register to default values).
I need to find out how configures or generates exceptions in FPU before the sample tasks.

PR #14830 (2019-03-22): arm: fix thread's FPU initialization
FPU's control and status register (FPCSR) for a thread is not initialized.
Random values are written to this register and this leads to failure on
fp_sharing test.
FPCSR register is set to 0 according to the value of FPDSCR.

Fixes: #14606 #9656
closed 2019-03-22 by galak