zephyrproject-rtos/zephyr #14609: mimxrt1050_evk Fatal fault in thread tests/kernel/mem_protect/stackprot Fatal fault in thread

hakehuang, 2019-03-18:
**Describe the bug**
kernel mem_protect stack prot

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=mimxrt1050_evk ..
3. make
4. See error

**Expected behavior**
test PASS

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Screenshots or console output**
```
***** Booting Zephyr OS v1.14.0-rc1-1369-gf30a1c03a260 *****

Starts alternate_thread

alternate_thread: Input string is too long and stack overflowed!

***** Stack Check Fail! *****

Current thread ID = 0x80000068

Faulting instruction address = 0x6000ae44

Fatal fault in thread 0x80000068! Aborting.

Running test suite stackprot

===================================================================

starting test - 1?:抋?x?jW焰}屠.

鳃?
糃?栳?b愞w唼?盛畈c诽F嗕籈vN?媮+8D3?物?孯PH!8y"t3?%g榥[S€<D哥\硉s赼.严/r?贿pI$畴f揗谞偤&ps稦?耋?dI虸╟绕€橛尌z劬??夜∪{?忟W博燘撴?畛?崗D硎:?o??燘 侤?

*"$[8F?L>縞嶊<肌庛
[a?娔?!?t鯜駛诋兾阺x瑆饡怙b???惞MG)圩)fⅵ訒€r$kx櫚倒E5爵A孉~氼6K簳瓞廞愯u旑J爥0?
***** MPU FAULT *****

  Instruction Access Violation

***** Hardware exception *****

Current thread ID = 0x800001ec

Faulting instruction address = 0xe9e8

Fatal fault in thread 0x800001ec! Aborting.

```

**Environment (please complete the following information):**
 - OS: (Linux )
 - Toolchain ( Zephyr SDK)
 - Commit SHA or Version used: v1.14.0_rc2

hakehuang, 2019-03-18:
@MaureenHelm

agansari, 2019-03-19:
@hakehuang i'm not able to reproduce this.
What commit are you on? I've checked out v1.14.0-rc2 tag, but get the following header when running the board:
***** Booting Zephyr OS v1.14.0-rc1-1238-g350f6f715632 *****

The same goes for #14606 but I was able to reproduce that.

agansari, 2019-03-22:
sorry @hakehuang i can't reproduce the MPU FAULT in this issue in rc2

MaureenHelm, 2019-03-22:
I can't reproduce the fault either. Here's what I'm seeing:
```
***** Booting Zephyr OS v1.14.0-rc1-1238-g350f6f715632 *****
Starts alternate_thread
alternate_thread: Input string is too long and stack overflowed!

***** Stack Check Fail! *****
Current thread ID = 0x80000068
Faulting instruction address = 0x6000ad20
Fatal fault in thread 0x80000068! Aborting.
Running test suite stackprot
===================================================================
starting test - test_stackprot
test_stackprot: Stack ok
test_stackprot: Stack ok
test_stackprot: Stack ok
test_stackprot: Stack ok
test_stackprot: Stack ok
test_stackprot: Stack ok
PASS - test_stackprot
===================================================================
Test suite stackprot succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL
```

hakehuang, 2019-03-28:
this is test setting issue. close it

closed 2019-03-28 by hakehuang