zephyrproject-rtos/zephyr #14621: BLE controller: Add support for Controller(SW)-based Privacy

aescolar, 2019-03-18:
For devices without address resolution acceleration in the LL hardware (like a nRF51), the current controller code does not support privacy (random addresses).
A method for resolving addresses in SW should be added for this kind of platforms.
This only needs to be added for the new split LL architecture (as the old will be deprecated soon).

MaureenHelm, 2019-07-10:
Will this be ready for the 2.0.0 release or should we push out to a future release?

aescolar, 2019-07-11:
@MaureenHelm Please wait a few days. We may still be able to manage it: The 2 developers working on it are right now out on holidays, but the work is quite far, and it is a new feature with low risk for existing ones.
I will move the milestone myself if a few days (after they are back from holidays) if it still looks like we won't make it.

PR #19863 (2019-12-13): Bluetooth: controller: SW deferred privacy
Enable deferred resolve of RPAs.
In case resolve of RPAs is time consuming, a new structure is introduced of known unknown RPAs.
When an RPA is passed for resolving the known unknown list is examined, if not in the list, i.e. a new
RPA is seen, it is put on the work queue for resolving. Once resolve operation is done the LLL is
signaled through a callback.

Fixes #14621
closed 2019-12-13 by carlescufi