zephyrproject-rtos/zephyr #14622: net: IPv6: malformed packet in fragmented echo reply

rmstoi, 2019-03-18:
Prerequisites: enable CONFIG_NET_IPV6_FRAGMENT=y

There seems to be an issue with IPv6 fragmentation. For example, testing native posix echo server:

`ping6 -I zeth 2001:db8::1 -s 1300 -c 1`

To echo request Zephyr sends echo reply with the malformed second fragment. Wireshark marks that packet as malformed and ping6 does not succeed.

finikorg, 2019-03-18:
I have this issue for my netusb samples, interesting that if I disable IPv6 fragmentation I can send 1500 bytes - headers, while when fragmentation is enabled I can not send even 1300 bytes.

rmstoi, 2019-03-18:
The malformed fragment seems to occur after second echo request. The first echo reply has wrong checksum and its second  fragment is missing.

[fragv6malform.pcapng.gz](https://github.com/zephyrproject-rtos/zephyr/files/2978650/fragv6malform.pcapng.gz)

Also, I've changed the amount of buffers:

```
-CONFIG_NET_PKT_RX_COUNT=16
-CONFIG_NET_PKT_TX_COUNT=16
-CONFIG_NET_BUF_RX_COUNT=64
-CONFIG_NET_BUF_TX_COUNT=64
+CONFIG_NET_PKT_RX_COUNT=160
+CONFIG_NET_PKT_TX_COUNT=160
+CONFIG_NET_BUF_RX_COUNT=640
+CONFIG_NET_BUF_TX_COUNT=640

```

jukkar, 2019-03-18:
> I have this issue for my netusb samples, interesting that if I disable IPv6 fragmentation I can send 1500 bytes - headers, while when fragmentation is enabled I can not send even 1300 bytes.

The IPv6 MTU is 1280 bytes, although in this case we could and should send 1500 bytes just fine. If you feel like it, please create issue for that.

finikorg, 2019-03-19:
> The IPv6 MTU is 1280 bytes, although in this case we could and should send 1500 bytes just fine. If you feel like it, please create issue for that.

Added Issue #14659 (feature request)

PR #14785 (2019-04-02): net/ipv6: Fix IPv6 fragment nexthdr field in case of no option headers
It has to work on both case, currently it was setting always hop-by-hop
next header which is obviously wrong but worked on ipv6_fragment test
since that one has only packets with optional headers (hop-by-hop in
that case has to be the first optional header).

Fixes #14622
commit 26a335c2 (2019-04-02): net/ipv6: Fix IPv6 fragment nexthdr field in case of no option headers

With or without options headers it has to work. Currently it was
setting always hop-by-hop next header which is obviously wrong but
worked on ipv6_fragment test since that one has only packets with
optional headers (hop-by-hop in that case has to be the first optional
header).

Fixes #14622
closed 2019-04-02 by nashif