zephyrproject-rtos/zephyr #14623: sanitycheck error when trying to run specific test

ulfalizer, 2019-03-18:
Not sure if I'm doing this the intended way, but anyway:

```
$ scripts/sanitycheck -p qemu_x86_64 -s tests/subsys/logging/log_msg/logging.log_msg
JOBS: 16
Cleaning output directory /home/ulf/z/sanity-out
Building testcase defconfigs...
1 tests selected, 91842 tests discarded due to filters
total complete:    0/   1   0%  failed:    0
1 of 1 tests passed with 0 warnings in 5 seconds
Traceback (most recent call last):
  File "scripts/sanitycheck", line 3385, in <module>
    main()
  File "scripts/sanitycheck", line 3374, in main
    ts.testcase_xunit_report(LAST_SANITY_XUNIT, duration)
  File "scripts/sanitycheck", line 2578, in testcase_xunit_report
    handler_time = "%s" %(goal.metrics["handler_time"])
KeyError: 'handler_time'
```

nashif, 2019-03-19:
hmm, can't reproduce this.

nashif, 2019-03-26:
@ulfalizer ping, you still able to reproduce this?

nashif, 2019-03-29:
cant reproduce and no additional details.

closed 2019-03-29 by nashif