zephyrproject-rtos/zephyr #14633: undefined reference to `mbedtls_debug_set_threshold' when MBEDTLS_DEBUG is enabled

laperie, 2019-03-18:
**Describe the bug**

In #14632, enable 

**To Reproduce**
Steps to reproduce the behavior:
1. in #14632, enable MBEDTLS_DEBUG
2. build 
3. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots or console output**
/opt/zephyr-sdk/i586-zephyr-elf/bin/../lib/gcc/i586-zephyr-elf/8.3.0/../../../../i586-zephyr-elf/bin/ld: libzephyr.a(sockets_tls.c.obj): in function `tls_init':
/home/laperie/workspace/zephyr/zephyr/subsys/net/lib/sockets/sockets_tls.c:289: undefined reference to `mbedtls_debug_set_threshold'
collect2: error: ld returned 1 exit status
zephyr/CMakeFiles/zephyr_prebuilt.dir/build.make:99: recipe for target 'zephyr/zephyr_prebuilt.elf' failed
make[2]: *** [zephyr/zephyr_prebuilt.elf] Error 1

**Environment (please complete the following information):**
Linux
0.10.1 SDK

rlubos, 2019-03-19:
Hi @laperie, I've seen this issue before. It seems that `make` does not rebuild mbedTLS in case Zephyr configuration is changed (in this case mbedTLS debug enabled). When I used make, I forced it to rebuild entire project with `make clean && make` after changing mbedTLS configuration.

Apparently this is not an issue when using `ninja` - it seems to understand better what needs to be rebuilt.

laperie, 2019-03-19:
> Hi @laperie, I've seen this issue before. It seems that `make` does not rebuild mbedTLS in case Zephyr configuration is changed (in this case mbedTLS debug enabled). When I used make, I forced it to rebuild entire project with `make clean && make` after changing mbedTLS configuration.

Cannot confirm that, sorry -- same symptoms no matter what you use to build the sample.  Problem is likely elsewhere...

jukkar, 2019-03-20:
I have not used public server like in the example as I do not have direct internet access from my test host. So I used simple python https server with self certified certificate, could that self cert be the culprit here?

rlubos, 2019-03-20:
@jukkar I suppose the above comment reffered to #14632?

jukkar, 2019-03-20:
Ah, sorry, I seemed to have commented wrong issue :)

jukkar, 2019-05-22:
I just tried this and did not see any issues -> closing.

closed 2019-05-22 by jukkar

hraftery, 2024-06-06:
Adding the mysterious `CONFIG_MBEDTLS_DEBUG_C` fixes it for me.