zephyrproject-rtos/zephyr #14640: ARC: tests/cmsis_rtos_v2/portability.cmsis_rtos_v2 fails on nsim_em and nsim_sem

andrewboie, 2019-03-18:
```
===================================================================
starting test - test_thread_apis

    Assertion failed at /home/apboie/projects/zephyr3/zephyr/tests/cmsis_rtos_v2/src/thread_apis.c:88: thread2: (osThreadGetStackSize(thread_array[i]) <= STACKSZ is false)
stack size allocated is not what is expected
FAIL - test_thread_apis
===================================================================
starting test - test_thread_apis_dynamic

    Assertion failed at /home/apboie/projects/zephyr3/zephyr/tests/cmsis_rtos_v2/src/thread_apis.c:134: thread_apis_common: (osThreadGetCount() not equal to 2)
Incorrect number of cmsis rtos v2 threads
FAIL - test_thread_apis_dynamic
===================================================================
starting test - test_thread_prio

    Assertion failed at /home/apboie/projects/zephyr3/zephyr/tests/cmsis_rtos_v2/src/thread_apis.c:63: thread1: *args->yield_check not equal to 2

FAIL - test_thread_prio
===================================================================
starting test - test_thread_prio_dynamic

    Assertion failed at /home/apboie/projects/zephyr3/zephyr/tests/cmsis_rtos_v2/src/thread_apis.c:50: thread1: (strcmp(args->name, name) == 0 is false)
Failed getting Thread name
FAIL - test_thread_prio_dynamic
===================================================================
starting test - test_timer

    Assertion failed at /home/apboie/projects/zephyr3/zephyr/tests/cmsis_rtos_v2/src/thread_apis.c:83: thread2: (num_threads not equal to 2)
Incorrect number of cmsis rtos v2 threads
FAIL - test_timer

```

vonhust, 2019-03-26:
This is caused by the mismatch in stack_info,  the same root cause of #13561 #14766. When fix #13561 #14766, this issue will be fixed.

nashif, 2019-04-02:
moving to low because blocking bug is low now.

andrewboie, 2019-04-02:
> This is caused by the mismatch in stack_info, the same root cause of #13561 #14766. When fix #13561 #14766, this issue will be fixed.

#13561 has nothing to do with this..

andrewboie, 2019-04-02:
> This is caused by the mismatch in stack_info, the same root cause of #13561 #14766. When fix #13561 #14766, this issue will be fixed.

And this has nothing to do with #14766 either.
The root cause is that the CMSIS test doesn't like that the stack generated is *larger* than the specified size. This isn't reasonable, it should complain if it is smaller.

closed 2019-04-03 by nashif

abrodkin, 2019-11-21:
Even with https://github.com/zephyrproject-rtos/zephyr/pull/15123 in place I still see a failure in `cmsis_rtos_v2` on nSIM with ARC EM:
```
***** Booting Zephyr OS build v2.1.0-rc1-215-gcb7974ab051f *****
Running test suite test_cmsis_v2_apis
===================================================================
starting test - test_kernel_apis
PASS - test_kernel_apis
===================================================================
starting test - test_delay
PASS - test_delay
===================================================================
starting test - test_thread_apis
E: ***** Exception vector: 0x6, cause code: 0x1, parameter 0x2
E: Address 0x80002a70
E: EV_ProtV
E: Memory read protection violation (stack checking scheme)
E: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
E: Current thread: 0x800001c4 (Thread2)
E: Halting system
```

FWIW I'm on today's master.
@vonhust do you see that issue still as well or it's just me doing something stupid?

vonhust, 2019-11-21:
@abrodkin  this test did operations on stack which is not allowed by STACK_CHECKING, plse see #17935 ,  two possible fixes:
* modify test code to remove operations on stack
* disable arc's STACK_CHECKING for this test.

PR #15123 (2019-04-03): tests: cmsis_rtos_v2: fix thread checks
It's really not possible to design a test where we can
enforce expectations on real vs. expected stack size:

- Some platforms may increase stack size over what is expected
  due to rounding up the stack buffer area to the next power
  of two.

- Some configuration options like CONFIG_STACK_RANDOM
  carve out space in the stack buffer, resulting in a stack
  size less than what is expected.

Best we can do is just assert that the amount of space
available should be less than the total size reported.

Fixes: #14640
commit 3f2d2f1d (2019-04-03): tests: cmsis_rtos_v2: fix thread checks

It's really not possible to design a test where we can
enforce expectations on real vs. expected stack size:

- Some platforms may increase stack size over what is expected
  due to rounding up the stack buffer area to the next power
  of two.

- Some configuration options like CONFIG_STACK_RANDOM
  carve out space in the stack buffer, resulting in a stack
  size less than what is expected.

Best we can do is just assert that the amount of space
available should be less than the total size reported.

Fixes: #14640