zephyrproject-rtos/zephyr #14641: ARC: tests/kernel/critical/kernel.common hangs on nsim_em

andrewboie, 2019-03-18:
Test case just hangs on nsim_em simulator:

```
***** Booting Zephyr OS v1.14.0-rc1-1250-g4e097b4d541f *****
Running test suite kernel_critical
===================================================================
starting test - test_critical
```

andrewboie, 2019-03-18:
It looks like the test eventually completes, but it takes a long time, longer than the allowed timeout in sanitycheck for the test.

On qemu_x86, the test takes about 16 seconds.
On nsim_em, the test takes about 105 seconds on my machine.

vonhust, 2019-03-20:
This may be fixed by #14558, where i used 10 Mhz  host timer as timer resource. 

If the timer is accurate,  the critical test will run 1500-1600 ms.

nashif, 2019-03-30:
this was already fixed by #14558

PR #14900 (2019-04-02): board: nsim_em: fix the bug in nsim host timer
* the nsim host timer does not work as expected,
  disable it, use cycle count to simulate timer tick

* optmize the freq definition of nsim_em

tests/kernel/critical test still may time out,  one possible fix is to increase its timeout value.

Fixes: #14641
closed 2019-04-02 by nashif