zephyrproject-rtos/zephyr #14642: ARC: tests/posix/common/ fails on nsim_sem

andrewboie, 2019-03-18:
Only seen failure on nsim_sem. Fails both test variants: portability.posix and portability.posix.newlib

The failure is a hang after printing some output:

```
***** Booting Zephyr OS v1.14.0-rc1-1250-g4e097b4d541f *****
Running test suite posix_apis
===================================================================
starting test - test_posix_pthread_execution
Thread 0 starting with scheduling policy 0 & priority 15
Thread 1 starting with scheduling policy 0 & priority 15
Thread 2 starting with scheduling policy 0 & priority 15

```

andrewboie, 2019-03-18:
It looks like the test actually succeeds, but takes a long time. On my unloaded workstation it took about 21 seconds for the test to complete.

andrewboie, 2019-04-02:
This seems to be worse since the last time I looked at it, it now appears to get stuck.

andrewboie, 2019-04-02:
Correction. On nsim_em, the test takes about 22 seconds. on nsim_sem, it hangs forever. The bug is only on nsim_sem.

andrewboie, 2019-04-02:
I think there is some weird race in the test itself, I'm digging deeper.

andrewboie, 2019-04-02:
the slowness of the test is due to the sched_yield() call in thread_top_exec(), which simply calls k_yield(), presumably to allow other threads to run...if I comment it out the test finishes very quickly.

Not sure what is going on yet.

andrewboie, 2019-04-02:
More data:

If I disable CONFIG_TEST_USERSPACE (which runs the test without user mode), it completes very quickly.

andrewboie, 2019-04-03:
tests/kernel/critical has a similar issue:

**Time to run on nsim_sem: 116s**
Time to run on nsim_em: 13s

Now if I comment out the body of configure_mpu_thread(), I get:

Time to run on nsim_sem: 11s
Time to run on nsim_em: 7s

andrewboie, 2019-04-03:
Both tests involve a lot of context switching, the performance of the MPU reprogramming (either in the code or the emulator) is severely impacting the test completion time. Closing the other bug #15129 as a duplicate.

vonhust, 2019-04-03:
@andrewboie , if i recall correctly, it runs ok in hardware. I will double check it

andrewboie, 2019-04-03:
Tomorrow I'm going to try to apply the optimization in #15135, it seems ARC code unconditionally reprograms all of the volatile MPU registers on every context switch

vonhust, 2019-04-03:
@andrewboie  no problem in hardware, both em and sem are ok.  The difference betwenn call configure_mpu_thread and not call configure_mpu_thread is time of context switch

ruuddw, 2019-04-04:
I did a quick experiment, looking at the count values/number of iterations completed in the (fixed) test time. There is (as can be expected) significant context switch 'overhead' due to the MPU reprogramming on both nSIM and hardware (~25%-100% less iterations with userspace enabled, for both MPU variants) but that doesn't explain the 10x slower performance on nsim_sem. Based on this, my expectation is nSIM simulation performance with the MPU "v3" is the cause of the timeouts, not the context switch overhead (though there is room for improvement for that as well). I'll check/follow up with the nSIM team.

andrewboie, 2019-04-06:
> that doesn't explain the 10x slower performance on nsim_sem

@ruuddw nsim_sem does tons more writes to the MPU registers than nsim_em does, to implement gap-filling. See details in #15223

PR #15274 (2019-04-09): tests: add nsim special cases for two tests
The NSIM emulator has severe performance issues when
the MPU registers are reprogrammed on context switch.
Disable runtime reprogramming of the MPU for these
platforms on these two tests, which have a lot of context
switch thrashing. This is done by ensuring userspace
and hardware stack overflow detection via guard areas
is disabled.

I have assurances from the ARC team that the tests run fine
on real hardware and this is an emulation issue. We're only
changing the test parameters for these emulated targets.

For 1.15, this will be completely resolve [... cut]
commit 7387b56c (2019-04-09): tests: add nsim special cases for two tests

The NSIM emulator has severe performance issues when
the MPU registers are reprogrammed on context switch.
Disable runtime reprogramming of the MPU for these
platforms on these two tests, which have a lot of context
switch thrashing. This is done by ensuring userspace
and hardware stack overflow detection via guard areas
is disabled.

I have assurances from the ARC team that the tests run fine
on real hardware and this is an emulation issue.

For 1.15, this will be completely resolved by optimizing
MPU region gap-filling to not take place during con [... cut]
closed 2019-04-09 by nashif