zephyrproject-rtos/zephyr #14654: Samples: echo_client: No reply packet from the server

tedd-an, 2019-03-19:
**Describe the bug**
echo_client sample is not working as described [here](https://docs.zephyrproject.org/latest/samples/net/sockets/echo_client/README.html).
The echo_client sends the packet but it is not received by the echo_server.

**To Reproduce**
Steps to reproduce the behavior:
1. cd samples/net/sockets/echo_client
2. mkdir build; cd build
3. cmake -GNinja -DBOARD=qemu_x86 ..
4. ninja

5. From the console, go to net-tools/
$ ./loop-socat.sh
$ sudo ./loop-slip-tap.sh
$ sudo ./echo-server -i tap0

6. ninja run
7. No packets between client and server

**Expected behavior**
echo-client should send and receives the packet

**Impact**
Sample is not working

**Screenshots or console output**
Attached the console output
[zephyr-output.txt](https://github.com/zephyrproject-rtos/zephyr/files/2980999/zephyr-output.txt)

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK
 - Commit SHA or Version used
HEAD 91eb1474522b766eb3294089fb251aff061aa1b4

**Additional context**
Add any other context about the problem here.

rlubos, 2019-03-19:
It's most likely an IP address mismatch - the default sample configuration allows for seamless Qemu vs Qemu, or board vs board communication. Qemu vs Linux requires to swap IP address in the Echo Client as mentioned in a comment in a project file https://github.com/zephyrproject-rtos/zephyr/blob/master/samples/net/sockets/echo_client/boards/qemu_x86.conf.

Probably we should add this information in the documentation as well, to avoid confusion in the future.

jukkar, 2019-03-19:
That is indeed the most probable case here. So the convention has been that zephyr is running in 192.0.2.1 or 2001:db8::1 and the Linux is in 192.0.2.2 or 2001:db8::2. Of course this is not valid if we have two zephyrs connected to each other, in that case the server should be in 192.0.2.1 and client in 192.0.2.2. There is nothing preventing to use any IP addresses but if we stick with these examples, then it is at least somewhat easier to do the testing.

tedd-an, 2019-03-19:
Confirmed that as @rlubos suggested, it works find with Qemu vs Linux mode after swapping the IP address.
However, the page contains invalid information:
[https://docs.zephyrproject.org/latest/samples/net/sockets/echo_client/README.html#running-echo-server-in-linux-host](https://docs.zephyrproject.org/latest/samples/net/sockets/echo_client/README.html#running-echo-server-in-linux-host)

It may be a good idea to update the page instead of fixing anything here.

PR #14690 (2019-03-20): Samples: echo-client: Add overlay when echo-server is running on Linux
This patch adds the overlay when the echo-server is running on Linux and
echo-client is running on Qemu.

Fix: #14654
commit 980cf393 (2019-03-20): Samples: echo-client: Add overlay when echo-server is running on Linux

This patch adds the overlay when the echo-server is running on Linux and
echo-client is running on Qemu.

Fix: #14654
closed 2019-03-20 by galak