zephyrproject-rtos/zephyr #14661: samples/net/syslog_net fails for native_posix

laperie, 2019-03-19:
**Describe the bug**
In native_posix configuration, syslog_net sample is not working

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=native_posix 
3. make run
4. Ensure syslogd is listening to the UDP port 514.
For example, in xubuntu, ensure that following is uncommented in /etc/rsyslogd.conf
``module(load="imudp")``
``input(type="imudp" port="514")``
5. Run
6. Observe syslog updates (e.g tail -f /var/log/syslog)

**Expected behavior**
Log items generated by Zephyr running in native_posix shoud be visible in syslog

**Impact**
While the sample works for qemu_x86 it does not generate any log entries for native_posix
Checking in Wireshark, reveals that no packets is sent over the net
 
**Environment (please complete the following information):**
Linux
SDK 0.10.0

jukkar, 2019-03-19:
I tested this yesterday and noticed that in this case the logger did not call the net backend output function. This looks like some glitch in logger to me.
Also it is enough to verify that the log messages are sent by just monitoring the `zeth` network interface in host using wireshark. Before starting zephyr, one needs to run net-setup.sh script from net-tools repo (it creates the `zeth` interface in host).

Assigning to @nordic-krch for verification if this is indeed issue in logger.

nordic-krch, 2019-03-19:
i suspect one thing. Immediate mode (former in_place mode) has been reworked and that required changes in backends. I guess that network backend was not updated. If i recall correctly native_posix board is forcing `LOG_IMMEDIATE` option and network backend simply does not support it.

To add that 2 functions are required, one for synchronous string processing and second one for hexdump. You can see the example here:
https://github.com/zephyrproject-rtos/zephyr/blob/2b1227f1551270e202781a9608490e1150c13b46/subsys/logging/log_backend_uart.c#L73

nordic-krch, 2019-03-19:
@jhedberg do you know if bluetooth monitor backend can be used in native posix? If yes then it will hit the same issue - missing support for working in blocking mode.

jhedberg, 2019-03-19:
> @jhedberg do you know if bluetooth monitor backend can be used in native posix? If yes then it will hit the same issue - missing support for working in blocking mode.

@nordic-krch I've never tried it, but seems a bit pointless since on native posix you get HCI traces straight from the linux kernel using btmon, i.e. no need to ask Zephyr to give them to you.

jukkar, 2019-03-19:
Ok, good to know. I can do the fixing for the backend and test it. Re-assigning back to me.

jukkar, 2019-03-19:
BTW, in the future, it would be good that if changes to backends are done, then all backends would be converted too, or at least inform the relevant parties that some changes are needed.

nordic-krch, 2019-03-19:
@jukkar i assumed that network backend will not work in synchronous mode because that means that it must support processing even from interrupts. I didn't take native_posix board into consideration.

jukkar, 2019-03-19:
@nordic-krch no worries, I have a fix coming in few minutes.

PR #14672 (2019-03-20): log: net: Fix the immediate logging mode in network backend
If CONFIG_LOG_IMMEDIATE is enabled, then we need to handle
the output strings using a different output function.
The issue was noticed with native_posix board where no syslog
output strings were sent to network.

Fixes #14661
commit 3f34fead (2019-03-20): log: net: Fix the immediate logging mode in network backend

If CONFIG_LOG_IMMEDIATE is enabled, then we need to handle
the output strings using a different output function.
The issue was noticed with native_posix board where no syslog
output strings were sent to network.

Fixes #14661
closed 2019-03-20 by galak