zephyrproject-rtos/zephyr #14668: net: icmp4: Zephyr strips record route and time stamp options

rmstoi, 2019-03-19:
If echo server receives ICMPv4 echo request with IPv4 options Record Route and Time Stamp it does not include them in echo response. According to [RFC 1122 3.2.2.6](https://tools.ietf.org/html/rfc1122#page-43)
 these options must be updated and included in the echo response.

[ipv4-timestamp-recordroute.pcap.gz](https://github.com/zephyrproject-rtos/zephyr/files/2982874/ipv4-timestamp-recordroute.pcap.gz)

tbursztyka, 2019-03-19:
didn't you say that linux was dropping the options, when I fixed the ipv4 options parsing?

tbursztyka, 2019-03-19:
oh btw, we don't support ANY ipv4 options. So it's not a bug, we just don't support any ipv4 options, period.

If we want to support it, that's an enhancement

rmstoi, 2019-03-19:
> didn't you say that linux was dropping the options, when I fixed the ipv4 options parsing?

Please add reference, to where I said that

tbursztyka, 2019-03-19:
@rmstoi It was on IRC. But that was about #11618 so probably not about these same options. nevermind

jukkar, 2019-05-22:
This is more like an enhancement than feature.

rmstoi, 2019-09-30:
pcap of this issue:

Linux DUT test PASS
[linux-ipv4-timestamp-recordroute.pcap.gz](https://github.com/zephyrproject-rtos/zephyr/files/3671426/linux-ipv4-timestamp-recordroute.pcap.gz)

Zephyr DUT PR #19470 test FAIL
[zephyr-pr-19470.pcap.gz](https://github.com/zephyrproject-rtos/zephyr/files/3671429/zephyr-pr-19470.pcap.gz)

FAIL reason is:
"FAIL: First Record Route entry is not DUT's IPv4 address. Time Stamp Route entry is not DUT's IPv4 address. "

PR #19470 (2019-12-16): Add support for IPv4 header options in EchoRequest
Timestamp and RecordRoute options are supported in only EchoRequest call as per RFC 1122 3.2.2.6.

Fixes #14668
commit 1c339513 (2019-12-16): net: icmpv4: Add support for Timestamp and RR

Timestamp and RecordRoute options are supported
in only ICMPv4 EchoRequest call as per
RFC 1122 3.2.2.6.

Fixes #14668
closed 2019-12-16 by jukkar