zephyrproject-rtos/zephyr #14678: Bluetooth: split controller add support for (current) Mesh Extensions

aescolar, 2019-03-19:
Add support to the split link layer for the controller mesh extensions which are already supported in the old link layer

jhedberg, 2019-04-24:
I'm not sure it's worth trying to do this for 2.0. There are no users (I know of at least) of these extensions with the old controller. I'd prefer if the full mesh vendor spec was implemented and then update the mesh stack to use them when available. However, if no-one commits to do that for 2.0 then just set some future milestone.

aescolar, 2019-04-25:
Thanks @jhedberg , milestone changed

taharvey, 2020-03-23:
Can you provide some color to the current status vs what needs to be built? What mesh features currently work in spilt controller mode? Does this also apply to Linux and BlueZ bridging? What needs completion?

aescolar, 2020-04-27:
@jhedberg @cvinayak ^^

jhedberg, 2020-04-27:
@taharvey all mesh features work with the current controller. This issue is about vendor HCI extensions that would optimize low-level packet filtering and improve transmission timing accuracy.

aescolar, 2023-03-23:
After talking with @cvinayak : There is no plan to implement vendor specific mesh extensions in the controller. This issue is "Closed as not planned"

closed 2023-03-23 by aescolar