zephyrproject-rtos/zephyr #14683: need end-to-end memory protection samples

andrewboie, 2019-03-19:
We need an end-to-end memory protection sample which demonstrates the following. We have ample tests for these but not an example showing everything working together.

- Multiple logical applications, each having their own threads, partitioned with memory domains
- Instantiation and assignment to a partition of a sys_mem_pool
- Configuration of thread resource pools
- Management of permission of drivers and kernel objects
- Allocation of dynamic kernel objects
- Definition of an application-specific system call

ioannisg, 2019-07-31:
@andrewboie I would like to know if you plan to work on this for 2.0, or whether we can move this to the following release.

andrewboie, 2019-07-31:
I am doing this for 2.0.

PR #19908 (2019-10-31): samples: add userspace producer/consumer sample
We try to demonstrate some concept for user mode:

- Multiple logical applications, each with their own memory
  domain
- Creation of a sys_mem_pool and assignment to a memory
  partition
- Use of APIs like k_queue_alloc_append() which require
  thread resource pools to be configured
- Management of permissions for kernel objects and drivers
- Show how application-specific system calls are defined
- Show IPC between ISR and application (using k_msgq) and
  application-to-application IPC (using k_queue)

Fixes: #14683
commit 6f928094 (2019-10-31): samples: add userspace producer/consumer sample

We try to demonstrate some concepts for user mode:

- Multiple logical applications, each with their own memory
  domain
- Creation of a sys_mem_pool and assignment to a memory
  partition
- Use of APIs like k_queue_alloc_append() which require
  thread resource pools to be configured
- Management of permissions for kernel objects and drivers
- Show how application-specific system calls are defined
- Show IPC between ISR and application (using k_msgq) and
  application-to-application IPC (using k_queue)

Fixes: #14683
closed 2019-10-31 by andrewboie