zephyrproject-rtos/zephyr #14688: remove privileged stack generation mechanism for ARM

andrewboie, 2019-03-19:
On ARM, we have logic in the build which scans the zephyr binary for thread objects and creates a gperf table mapping thread objects to separately declared supervisor mode stacks.

ARC has show us that this isn't necessary. We do not need to program MPU regions for the supervisor mode stacks, only user mode stacks. Even on MPU power of two aligned systems, the supervisor mode stack just needs to be aligned for its minimally sized guard region.

Remove the privilege stack generation logic in favor of ARC's much simpler solution, which is just to allocate the supervisor mode stack memory at the end of the stack buffer, after the user stack.

ioannisg, 2019-03-19:
@andrewboie @agross-linaro , if we do this, I would recommend to think of placing the priv stack lower than the use stack, if possible. So we could skip reprogramming the stack overflow limit at each system call (Armv8m)

andrewboie, 2019-03-19:
> I would recommend to think of placing the priv stack lower than the use stack, if possible

Given that the user stack may need to be aligned to its power-of-two size, I don't see how that is possible. This only works if the supervisor stack is at higher memory addresses.

andrewboie, 2019-04-09:
I'm going to close this, as the ARC approach has issues with alignment. Separate privilege mode stacks are going to need to stay for power-of-two systems.

closed 2019-04-09 by andrewboie