zephyrproject-rtos/zephyr #14691: samples: telnet: net shell is not working

tedd-an, 2019-03-19:
**Describe the bug**
Found couple of issues:
log_strdup_alloc_failed error
net shell is not working from the telnet side even if the connection is established

**To Reproduce**
Steps to reproduce the behavior:
1. samples/net/telnet
2. mkdir build; cd build
3. cmake -GNinja -DBOARD=qemu_x86 ..
4. ninja && ninja run
5. From the terminal, 
$ telnet 192.0.2.1

**Expected behavior**
Access net shell via telnet

**Impact**
Sample is not working as described in [https://docs.zephyrproject.org/latest/samples/net/telnet/README.html](https://docs.zephyrproject.org/latest/samples/net/telnet/README.html)

**Screenshots or console output**
output attached
[telnet-output.txt](https://github.com/zephyrproject-rtos/zephyr/files/2985774/telnet-output.txt)

**Environment (please complete the following information):**
 - OS: Linux FC28
 - Toolchain Zephyr SDK
 - 3dea408405fd8958f94ef24ab5d5251d6288841c

tedd-an, 2019-03-19:
it looks like similar issue #12611

jukkar, 2019-03-20:
Yes, the telnet cannot be used atm as the shell is missing support for network transport.

jukkar, 2019-03-20:
Duplicate of #12611

closed 2019-03-20 by jukkar