zephyrproject-rtos/zephyr #14715: kernel/mem_slab.c: Improve code coverage to 100%

cinlyooi-intel, 2019-03-20:
Improve code coverage for this file. At present, the code coverage command below reports the following:

Commit ID: 3dea408405fd8958f94ef24ab5d5251d6288841c
Line coverage ( %  hit/total): 100% 50/50
function coverage ( %  hit/total): 100% 5/5
branch coverage ( %  hit/total): 87.5% 14/16

```
./scripts/sanitycheck --coverage --coverage-platform mps2_an385 -p mps2_an385 --coverage-platform qemu_x86 -p qemu_x86 --coverage-platform qemu_x86_nommu -p qemu_x86_nommu --gcov-tool /opt/zephyr-sdk-0.10.0/sysroots/x86_64-pokysdk-linux/usr/bin/i586-zephyr-elf/i586-zephyr-elf-gcov -T tests/kernel -T tests/subsys/logging
```

cinlyooi-intel, 2019-03-20:
I believe we can close this one since it is the __ASSERT(condition) condition==true branch that were not covered

aescolar, 2019-03-20:
@cinlyooi-intel : You may want to have LCOV exclusions in the code for statements which are not meant to be covered (as self documentation of the exclusions). That way the coverage will also be reported as 100% .

cinlyooi-intel, 2019-03-21:
@aescolar : Thanks for the heads up. One reason for creating this ticket is I was not 100% certain that I can ignore all __ASSERT() (and Z_OOPS()), particularly those with complex branching arrangement.

closed 2019-04-18 by nashif