zephyrproject-rtos/zephyr #14749: Verify all samples work as intended

nashif, 2019-03-20:
- [ ] application_development/external_lib
 - [ ] application_development/out_of_tree_board
 - [ ] application_development/code_relocation
 - [ ] crypto/mbedtls_benchmark
 - [ ] mpu/mpu_test
 - [ ] mpu/mpu_stack_guard_test
 - [ ] mpu/mem_domain_apis_test
 - [ ] net/syslog_net
 - [ ] net/promiscuous_mode
 - [ ] net/wifi
 - [ ] net/dhcpv4_client
 - [ ] net/sockets/can
 - [ ] net/sockets/sntp_client
 - [ ] net/sockets/echo
 - [ ] net/sockets/echo_async_select
 - [ ] net/sockets/coap_server
 - [ ] net/sockets/echo_client
 - [ ] net/sockets/echo_async
 - [ ] net/sockets/http_get
 - [ ] net/sockets/big_http_download
 - [ ] net/sockets/dumb_http_server
 - [ ] net/sockets/coap_client
 - [ ] net/sockets/echo_server
 - [ ] net/sockets/packet
 - [ ] net/mdns_responder
 - [ ] net/wpanusb
 - [ ] net/lldp
 - [ ] net/mqtt_publisher
 - [ ] net/eth_native_posix
 - [ ] net/vlan
 - [ ] net/nats
 - [ ] net/zperf
 - [ ] net/gptp
 - [ ] net/lwm2m_client
 - [ ] net/stats
 - [ ] net/wpan_serial
 - [ ] net/ipv4_autoconf
 - [ ] net/telnet
 - [ ] net/dns_resolve
 - [ ] net/google_iot_mqtt
 - [ ] drivers/led_ws2812
 - [ ] drivers/led_lp5562
 - [ ] drivers/crypto
 - [ ] drivers/CAN
 - [ ] drivers/flash_shell
 - [ ] drivers/led_lpd8806
 - [ ] drivers/led_apa102
 - [ ] drivers/entropy
 - [ ] drivers/i2c_scanner
 - [ ] drivers/spi_fujitsu_fram
 - [ ] drivers/watchdog
 - [ ] drivers/led_pca9633
 - [ ] drivers/rtc
 - [ ] drivers/led_lp3943
 - [ ] drivers/led_apa102c_bitbang
 - [ ] drivers/i2c_fujitsu_fram
 - [ ] drivers/soc_flash_nrf
 - [ ] drivers/lcd_hd44780
 - [ ] drivers/counter/alarm
 - [ ] drivers/current_sensing
 - [ ] drivers/spi_flash
 - [ ] drivers/gpio
 - [ ] hello_world
 - [ ] synchronization
 - [ ] shields/x_nucleo_iks01a1
 - [ ] shields/x_nucleo_iks01a2
 - [ ] subsys/power/device_pm
 - [ ] subsys/mgmt/mcumgr/smp_svr
 - [ ] subsys/shell/shell_module
 - [ ] subsys/usb/mass
 - [ ] subsys/usb/dfu
 - [ ] subsys/usb/hid-cdc
 - [ ] subsys/usb/hid-mouse
 - [ ] subsys/usb/cdc_acm
 - [ ] subsys/usb/testusb
 - [ ] subsys/usb/cdc_acm_composite
 - [ ] subsys/usb/hid
 - [ ] subsys/usb/console
 - [ ] subsys/usb/webusb
 - [ ] subsys/ipc/openamp
 - [ ] subsys/ipc/ipm_mcux
 - [ ] subsys/ipc/ipm_imx
 - [ ] subsys/ipc/ipm_mailbox/ap
 - [ ] subsys/ipc/ipm_mailbox/sensor
 - [ ] subsys/fs
 - [ ] subsys/logging/logger
 - [ ] subsys/nvs
 - [ ] subsys/console/echo
 - [ ] subsys/console/getchar
 - [ ] subsys/console/getline
 - [ ] basic/blinky
 - [ ] basic/blink_led
 - [ ] basic/disco
 - [ ] basic/servo_motor
 - [ ] basic/button
 - [ ] basic/threads
 - [ ] basic/rgb_led
 - [ ] basic/fade_led
 - [ ] cpp_synchronization
 - [ ] nfc/nfc_hello
 - [ ] sensor/lsm303dlhc
 - [ ] sensor/sht3xd
 - [ ] sensor/amg88xx
 - [ ] sensor/sx9500
 - [ ] sensor/grove_light
 - [ ] sensor/lsm6dsl
 - [ ] sensor/magn_polling
 - [ ] sensor/mcp9808
 - [ ] sensor/ms5837
 - [ ] sensor/thermometer
 - [ ] sensor/max30101
 - [ ] sensor/fxos8700
 - [ ] sensor/vl53l0x
 - [ ] sensor/bme280
 - [ ] sensor/hts221
 - [ ] sensor/bmg160
 - [ ] sensor/apds9960
 - [ ] sensor/grove_temperature
 - [ ] sensor/fxas21002
 - [ ] sensor/max44009
 - [ ] sensor/adt7420
 - [ ] sensor/bmm150
 - [ ] sensor/bmi160
 - [ ] sensor/fxos8700-hid
 - [ ] sensor/tmp112
 - [ ] sensor/adxl372
 - [ ] sensor/ccs811
 - [ ] sensor/th02
 - [ ] bluetooth/mesh
 - [ ] bluetooth/hci_usb
 - [ ] bluetooth/hci_uart
 - [ ] bluetooth/hci_spi
 - [ ] bluetooth/peripheral_csc
 - [ ] bluetooth/peripheral_hids
 - [ ] bluetooth/peripheral_dis
 - [ ] bluetooth/ibeacon
 - [ ] bluetooth/handsfree
 - [ ] bluetooth/eddystone
 - [ ] bluetooth/peripheral_sc_only
 - [ ] bluetooth/peripheral_esp
 - [ ] bluetooth/peripheral
 - [ ] bluetooth/central
 - [ ] bluetooth/mesh_demo
 - [ ] bluetooth/scan_adv
 - [ ] bluetooth/central_hr
 - [ ] bluetooth/ipsp
 - [ ] bluetooth/peripheral_hr
 - [ ] bluetooth/beacon
 - [ ] boards/reel_board/mesh_badge
 - [ ] boards/intel_s1000_crb/audio
 - [ ] boards/intel_s1000_crb/i2s
 - [ ] boards/altera_max10/pio
 - [ ] boards/olimex_stm32_e407/ccm
 - [ ] boards/up_squared/gpio_counter
 - [ ] boards/96b_argonkey/sensors
 - [ ] boards/96b_argonkey/microphone
 - [ ] boards/bbc_microbit/pong
 - [ ] boards/bbc_microbit/display
 - [ ] boards/bbc_microbit/sound
 - [ ] boards/arduino_101/environmental_sensing/ap
 - [ ] boards/arduino_101/environmental_sensing/sensor
 - [ ] boards/nrf52/mesh/onoff-app
 - [ ] boards/nrf52/mesh/onoff_level_lighting_vnd_app
 - [ ] boards/nrf52/power_mgr
 - [ ] gui/lvgl
 - [ ] userspace/shared_mem
 - [ ] portability/cmsis_rtos_v1/timer_synchronization
 - [ ] portability/cmsis_rtos_v1/philosophers
 - [ ] portability/cmsis_rtos_v2/timer_synchronization
 - [ ] portability/cmsis_rtos_v2/philosophers
 - [ ] display/grove_display
 - [ ] display/cfb_shell
 - [ ] display/ili9340
 - [ ] display/cfb_custom_font
 - [ ] display/cfb
 - [ ] philosophers

nashif, 2019-03-20:
@laperie fyi

cinlyooi-intel, 2019-03-28:
Updated the list on 20190328

laperie, 2019-03-28:
@erwango , we have a spreadsheet https://docs.google.com/spreadsheets/d/1peGvQQg2gXogputGtR0DrUNLDM1FWqR3m7bamueRSSE/edit#gid=0 that has more details, for networking and bluetooth at least. I think for 1.14 you can use that one too.

mike-scott, 2019-03-28:
@nashif I marked lwm2m_client as tested last week in the spreadsheet, but its not reflecting here?

tautologyclub, 2019-04-09:
I can attest that `boards/nrf52/power_mgr` seems completely broken (tested on `nrf52840_pca10056`, with some RTTLogger configs added to prj.conf):

```
***** Booting Zephyr OS v1.14.0-rc3 *****

*** Power Management Demo on nrf52840_pca10056 ***
Demo Description
Application creates idleness, due to which System Idle Thread is
scheduled and it enters into various Low Powe
<-- App doing busy wait for 10 Sec -->

<-- App going to sleep for 10 Sec -->
[00:00:10.027,160] <err> power: Not enough time for PM operations: 999

<-- App doing busy wait for 10 Sec -->

<-- App going to sleep for 30 Sec -->
--> Entering to SYS_POWER_STATE_SLEEP_3 state.
--> Exited from SYS_POWER_STATE_SLEEP_3 state.
[00:00:59.935,424] <err> soc: Unsupported power state 1

<-- Disabling SYS_POWER_STATE_SLEEP_3 state --->

<-- App doing busy wait for 10 Sec -->

<-- App going to sleep for 10 Sec -->
[00:01:09.935,638] <err> soc: Unsupported power state 1
[00:01:09.982,818] <err> power: Not enough time for PM operations: 997
```

I know I got it to running as expected a while ago (maybe November-December?), perhaps I should start a new issue though.

closed 2020-03-11 by nashif