zephyrproject-rtos/zephyr #14766: K_THREAD_STACK_BUFFER() is broken

andrewboie, 2019-03-20:
This macro has a design flaw: it assumes that the offset to the stack buffer, within the stack object, is constant.

However, on ARC this offset changes depending on whether the stack is being used in supervisor or user mode. There are no reverse references to the thread in the stack object (nor is it possible to add one); this API is broken by design.

This API needs to be removed; if someone needs to know the bounds of an in-use stack, this information is available in the thread struct anyway.

nashif, 2019-03-29:
what are the plans for this issue? Is it going to be fixed for 1.14 release?

andrewboie, 2019-03-29:
Implementation is in progress.

andrewboie, 2019-03-29:
Setting to low, in current code this may not reveal the true stack buffer, but does not point to memory outside of the stack object.

PR #15195 (2019-04-05): kernel: demote K_THREAD_STACK_BUFFER() to private
This macro is slated for complete removal, as it's not possible
on arches with an MPU stack guard to know the true buffer bounds
without also knowing the runtime state of its associated thread.

As removing this completely would be invasive to where we are
in the 1.14 release, demote to a private kernel Z_ API instead.
The current way that the macro is being used internally will
not cause any undue harm, we just don't want any external code
depending on it.

The final work to remove this (and overhaul stack specification in
general) will take p [... cut]
commit 4e5c093e (2019-04-05): kernel: demote K_THREAD_STACK_BUFFER() to private

This macro is slated for complete removal, as it's not possible
on arches with an MPU stack guard to know the true buffer bounds
without also knowing the runtime state of its associated thread.

As removing this completely would be invasive to where we are
in the 1.14 release, demote to a private kernel Z_ API instead.
The current way that the macro is being used internally will
not cause any undue harm, we just don't want any external code
depending on it.

The final work to remove this (and overhaul stack specification in
general) will take  [... cut]
closed 2019-04-05 by nashif