zephyrproject-rtos/zephyr #14779: stm32: If the memory usage is high, the flash is abnormal.

KwonTae-young, 2019-03-21:
**Describe the bug**
If the memory usage on the STM32 board is high, flash will behave abnormally.
If you do `flash`, the board will not do anything.

**To Reproduce**
The test was `stm32f4_disco`.
I modified `hello_world` slightly to increase memory usage.

Steps to reproduce the behavior:
```
west init zephyrproject -m https://github.com/KwonTae-young/zephyr/ --mr memory_usage_flash
cd zephyrproject/zephyr
source zephyr-env.sh
mkdir samples/hello_world/build
cd samples/hello_world/build/
cmake -DBOARD=stm32f4_disco ..
make -j8 flash
```

**Expected behavior**
I expected to see `hello world` in console.

**Impact**
If memory usage is high, flash will behave abnormally.

**Screenshots or console output**
Memory usage is `83.59%`.
No errors occur during build and flash.
[image]

```
root:build# pwd
/tmp/zephyrproject/zephyr/samples/hello_world/build
root:build# 
root:build# cmake -DBOARD=stm32f4_disco ..
Zephyr version: 1.14.0
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.7", minimum required is "3.4") 
-- Selected BOARD stm32f4_disco
-- Found west: /usr/local/bin/west (found suitable version "0.5.6", minimum required is "0.5.4")
-- Loading /tmp/zephyrproject/zephyr/boards/arm/stm32f4_disco/stm32f4_disco.dts as base
-- Overlaying /tmp/zephyrproject/zephyr/dts/common/common.dts
Parsing Kconfig tree in /tmp/zephyrproject/zephyr/Kconfig
Loading /tmp/zephyrproject/zephyr/boards/arm/stm32f4_disco/stm32f4_disco_defconfig as base
Merging /tmp/zephyrproject/zephyr/samples/hello_world/prj.conf
Configuration written to '/tmp/zephyrproject/zephyr/samples/hello_world/build/zephyr/.config'
-- Cache files will be written to: /root/.cache/zephyr
-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Performing Test toolchain_is_ok
-- Performing Test toolchain_is_ok - Success
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/zephyrproject/zephyr/samples/hello_world/build
root:build# 
root:build# 
root:build# make -j8 flash
Scanning dependencies of target driver_validation_h_target
Scanning dependencies of target syscall_macros_h_target
Scanning dependencies of target kobj_types_h_target
Scanning dependencies of target syscall_list_h_target
[  2%] Generating include/generated/syscall_macros.h
[  2%] Generating include/generated/driver-validation.h
[  3%] Generating include/generated/kobj-types-enum.h, include/generated/otype-to-str.h
[  4%] Generating misc/generated/syscalls.json
[  4%] Built target syscall_macros_h_target
[  4%] Generating include/generated/syscall_dispatch.c, include/generated/syscall_list.h
[  4%] Built target driver_validation_h_target
[  4%] Built target kobj_types_h_target
[... 217 lines cut ...]
`stm32f4_disco` reduces the memory usage to 63.25%, and the flash works normally.
Below is the console output by memory usage in `stm32f4_disco`.
 *  SIZE = 30000(63.25%)
```
***** Booting Zephyr OS zephyr-v1.13.0-5942-g1f6ab221f242 *****
Hello World! stm32f4_disco
0x20000008
sizeof(test) = 120000
```

 * SIZE = 40000(83.59%)
```

```

3. Flash with ST-Link Utility
I tried to flash zephyr.bin generated by build using `ST-Link Utility`.
The result was the same.
[image]
```

```

I'm sorry I do not have enough English.
Thank you.

erwango, 2019-03-21:
Had a quick test and it seems this not systematic on STM32F4:
```
***** Booting Zephyr OS v1.14.0-rc1-1287-g2320973fea31 *****
Hello World! nucleo_f401re
0x20000008                                                                      
sizeof(test) = 92000   
```
With:
```
Memory region         Used Size  Region Size  %age Used
           FLASH:       12616 B       512 KB      2.41%
            SRAM:       96200 B        96 KB     97.86%
        IDT_LIST:         184 B         2 KB      8.98%
```

erwango, 2019-03-21:
Testing with stm32f4_board, I confirm the issue.
When flashing with anything larger than 83.59% of the RAM, STLink generates the following message that can be found in FAIL.txt on mounted drive:
`The application file format is unknown and cannot be parsed and/or processed`

I'll have a check with ST-Link expert.

KwonTae-young, 2019-03-22:
@erwango
Thank you for your prompt confirmation. :smile:

erwango, 2019-03-22:
@KwonTae-young, ok, fiixed, cf #14838:

```
Memory region         Used Size  Region Size  %age Used
           FLASH:       14092 B         1 MB      1.34%
             CCM:          0 GB        64 KB      0.00%
            SRAM:      130752 B       128 KB     99.76%
        IDT_LIST:         184 B         2 KB      8.98%
```
Result:
```
***** Booting Zephyr OS v1.14.0-rc1-1369-ge71ff9f962df *****                    
Hello World! stm32f4_disco                                                      
0x20000008                                                                      
sizeof(test) = 126400  
```

KwonTae-young, 2019-03-25:
@erwango
It works well.(SIZE=30000)

**Before**
```
Memory region         Used Size  Region Size  %age Used
           FLASH:       14040 B         1 MB      1.34%
            SRAM:      124352 B       192 KB     63.25%
        IDT_LIST:         184 B         2 KB      8.98%
```

**After**
```
Memory region         Used Size  Region Size  %age Used
           FLASH:       14092 B         1 MB      1.34%
             CCM:          0 GB        64 KB      0.00%
            SRAM:      124352 B       128 KB     94.87%
        IDT_LIST:         184 B         2 KB      8.98% 
```

PR #14838 (2019-03-26): stm32f407Xg: RAM is 128K + 64K CCM
stm32f407Xg package was defined with 192K RAM + 64K CCM.
This is wrong as total RAM is 192, including 64K CCM.

Fix RAM size definition

Fixes #14779
commit d098c600 (2019-03-26): dts: stm32: Fix erroneous RAM sizes when CCM is available

On various stm32 soc packages CCM is available and SRAM size
mentionned in datasheet include CCM.
Though, actual SRAM size defined in dtsi files should not
include CCM sizes.
Fix this on impacted dtsi files.

Fixes #14779
closed 2019-03-26 by nashif