zephyrproject-rtos/zephyr #14786: Information about old sdk version provides wrong download link

Mierunski, 2019-03-21:
**Describe the bug**
When trying to build with old sdk version, there is a message to update sdk.
Link provided in that message is not working

**To Reproduce**
Build something with older sdk version

**Expected behavior**
Correct link provided

**Impact**
annoyance

**Screenshots or console output**
```
Zephyr version: 1.14.0
-- Selected BOARD nrf52840_pca10056
CMake Error at /home/mimi/ncs/zephyr/cmake/toolchain/zephyr/host-tools.cmake:36 (message):
  The SDK version you are using is too old, please update your SDK.

  You need at least SDK version 0.10.0.

  You have version 0.9.5 (/home/mimi/opt/zephyr-sdk).

  The new version of the SDK can be downloaded from:

  
  https://github.com/zephyrproject-rtos/meta-zephyr-sdk/releases/download/0.10.0/zephyr-sdk-0.10.0-setup.run

Call Stack (most recent call first):
  /home/mimi/ncs/zephyr/cmake/host-tools.cmake:1 (include)
  /home/mimi/ncs/zephyr/cmake/app/boilerplate.cmake:379 (include)
  CMakeLists.txt:3 (include)

-- Configuring incomplete, errors occurred!
See also "/home/mimi/ncs/zephyr/build/CMakeFiles/CMakeOutput.log".
See also "/home/mimi/ncs/zephyr/build/CMakeFiles/CMakeError.log".
FAILED: build.ninja 
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK

jukkar, 2019-03-21:
Will be fixed by #14778

galak, 2019-03-21:
Fixed with merge of #14778

closed 2019-03-21 by galak