zephyrproject-rtos/zephyr #14791: Notification callbacks should have conn as argument

pdunaj, 2019-03-21:
**Is your enhancement proposal related to a problem? Please describe.**
Without conn as an argument it is impossible to tell which connection callback refers to. This is a problem for applications with multiple simultaneous connections.

This was discussed already on https://github.com/zephyrproject-rtos/zephyr/issues/13316.

Needed by use cases A3, B2 from https://github.com/zephyrproject-rtos/zephyr/issues/7473#issuecomment-388741527.

**Describe the solution you'd like**
`conn` argument in callbacks.

**Describe alternatives you've considered**
N/A

**Additional context**
See description

Vudentz, 2019-03-22:
Doesn't BT_GATT_CCC_MANAGED fixed that? I mean both their write and match callbacks do have a bt_conn so you should be able to detect which write to a CCC and/or match a configuration, Id think you probably just need the latter so you can just pass NULL to bt_gatt_notify and then use the return to filter what connections shall receive the notification.

Vudentz, 2019-04-02:
@pdunaj Any update on this matter?

Vudentz, 2019-05-02:
@pdunaj Im closing this one since there hasn't been any updates, please reopen if you think it is still a problem after the introduction of BT_GATT_CCC_MANAGED.

closed 2019-05-02 by Vudentz