zephyrproject-rtos/zephyr #14806: Assorted pylint warnings in Python scripts

ulfalizer, 2019-03-21:
Most of these are probably either too nitty or too broad to turn into failures in CI. I'll dump them here for reference, in case someone wants to fix something. I've pruned the list a bit (no hanging-indent yada yada spam and the like).

```
arch/x86/gen_mmu_x86.py:133:4: R0201: Method could be a function (no-self-use)
arch/x86/gen_mmu_x86.py:137:4: R0201: Method could be a function (no-self-use)
arch/x86/gen_mmu_x86.py:141:4: R0201: Method could be a function (no-self-use)
arch/x86/gen_mmu_x86.py:202:0: W0301: Unnecessary semicolon (unnecessary-semicolon)
arch/x86/gen_mmu_x86.py:413:38: R1716: Simplify chained comparison between the operands (chained-comparison)
arch/x86/gen_mmu_x86.py:478:8: C0200: Consider using enumerate instead of iterating with range and len (consider-using-enumerate)
arch/x86/gen_mmu_x86.py:484:24: R1716: Simplify chained comparison between the operands (chained-comparison)
arch/x86/gen_mmu_x86.py:485:24: R1716: Simplify chained comparison between the operands (chained-comparison)
boards/xtensa/intel_s1000_crb/support/create_board_img.py:67:0: W0301: Unnecessary semicolon (unnecessary-semicolon)
boards/xtensa/intel_s1000_crb/support/create_board_img.py:68:0: W0301: Unnecessary semicolon (unnecessary-semicolon)
boards/xtensa/intel_s1000_crb/support/create_board_img.py:69:0: W0301: Unnecessary semicolon (unnecessary-semicolon)
boards/xtensa/intel_s1000_crb/support/create_board_img.py:71:0: W0301: Unnecessary semicolon (unnecessary-semicolon)
doc/conf.py:115:83: C0326: No space allowed before :
doc/conf.py:414:42: C0326: No space allowed before bracket
doc/conf.py:450:0: W0311: Bad indentation. Found 3 spaces, expected 4 (bad-indentation)
doc/conf.py:451:0: W0311: Bad indentation. Found 3 spaces, expected 4 (bad-indentation)
doc/extensions/zephyr/application.py:266:61: W0613: Unused argument 'compact' (unused-argument)
doc/extensions/zephyr/html_redirects.py:47:31: W0613: Unused argument 'docname' (unused-argument)
doc/extensions/zephyr/link-roles.py:17:7: C1801: Do not use `len(SEQUENCE)` to determine if a sequence is empty (len-as-condition)
doc/extensions/zephyr/link-roles.py:33:4: W0102: Dangerous default value [] as argument (dangerous-default-value)
doc/extensions/zephyr/link-roles.py:33:4: W0102: Dangerous default value {} as argument (dangerous-default-value)
doc/scripts/fix_tex.py:13:0: R1711: Useless return at end of function or method (useless-return)
samples/net/google_iot_mqtt/src/private_info/create_keys.py:100:0: W0301: Unnecessary semicolon (unnecessary-semicolon)
scripts/ci/get_modified_boards.py:59:26: C0326: Exactly one space required after comma
scripts/ci/get_modified_tests.py:56:26: C0326: Exactly one space required after comma
scripts/dts/extract/clocks.py:105:40: W0640: Cell variable clock_cells_string defined in loop (cell-var-from-loop)
scripts/dts/extract/clocks.py:106:40: W0640: Cell variable clock_cell_name defined in loop (cell-var-from-loop)
scripts/dts/extract/clocks.py:107:40: W0640: Cell variable index defined in loop (cell-var-from-loop)
scripts/dts/extract/clocks.py:130:20: W0702: No exception type(s) specified (bare-except)
scripts/dts/extract/clocks.py:18:51: W0613: Unused argument 'def_label' (unused-argument)
scripts/dts/extract/clocks.py:95:40: W0640: Cell variable clock_cells_string defined in loop (cell-var-from-loop)
scripts/dts/extract/clocks.py:96:40: W0640: Cell variable index defined in loop (cell-var-from-loop)
scripts/dts/extract/default.py:22:4: W0221: Parameters differ from overridden 'extract' method (arguments-differ)
scripts/dts/extract/directive.py:12:0: R0205: Class 'DTDirective' inherits from object, can be safely removed from bases in python3 (useless-object-inheritance)
scripts/dts/extract/flash.py:53:20: C0201: Consider iterating the dictionary directly instead of calling .keys() (consider-iterating-dictionary)
scripts/dts/extract/globals.py:388:63: W0640: Cell variable label defined in loop (cell-var-from-loop)
scripts/dts/extract/globals.py:470:67: W0640: Cell variable label defined in loop (cell-var-from-loop)
scripts/dts/extract/interrupts.py:24:4: W0221: Parameters differ from overridden 'extract' method (arguments-differ)
scripts/dts/extract/interrupts.py:72:60: W0640: Cell variable l_cell_name defined in loop (cell-var-from-loop)
scripts/dts/extract/reg.py:25:4: W0221: Parameters differ from overridden 'extract' method (arguments-differ)
scripts/dts/extract/reg.py:90:15: C1801: Do not use `len(SEQUENCE)` to determine if a sequence is empty (len-as-condition)
scripts/dts/extract_dts_includes.py:78:9: R1714: Consider merging these comparisons with "in" to "prop in ('interrupts', 'interrupts-extended')" (consider-using-in)
scripts/elf_helper.py:514:0: W0301: Unnecessary semicolon (unnecessary-semicolon)
scripts/filter-known-issues.py:212:19: C0121: Comparison to True should be just 'expr' or 'expr is True' (singleton-comparison)
scripts/filter-known-issues.py:82:8: W1201: Specify string format arguments as logging function parameters (logging-not-lazy)
scripts/gen_app_partitions.py:134:34: C0326: Exactly one space required after comma
scripts/gen_gcov_files.py:54:12: W0702: No exception type(s) specified (bare-except)
scripts/gen_kobject_list.py:172:22: R1716: Simplify chained comparison between the operands (chained-comparison)
scripts/gen_kobject_list.py:276:11: R1714: Consider merging these comparisons with "in" to "kobj in ('device', '_k_thread_stack_element')" (consider-using-in)
scripts/gen_priv_stacks.py:76:36: W0613: Unused argument 'static_begin' (unused-argument)
scripts/gen_priv_stacks.py:76:50: W0613: Unused argument 'static_end' (unused-argument)
scripts/gitlint/zephyr_commit_rules.py:113:11: C1801: Do not use `len(SEQUENCE)` to determine if a sequence is empty (len-as-condition)
scripts/gitlint/zephyr_commit_rules.py:16:-1: W0105: String statement has no effect (pointless-string-statement)
scripts/kconfig/checkconfig.py:112:28: C0121: Comparison to True should be just 'expr' or 'expr is True' (singleton-comparison)
scripts/kconfig/checkconfig.py:118:28: C0121: Comparison to False should be 'not expr' or 'expr is False' (singleton-comparison)
scripts/kconfig/checkconfig.py:124:8: C0121: Comparison to True should be just 'expr' or 'expr is True' (singleton-comparison)
scripts/kconfig/checkconfig.py:141:4: C0326: Exactly one space required before assignment
scripts/kconfig/checkconfig.py:147:48: C0326: Exactly one space required after comma
scripts/kconfig/checkconfig.py:19:9: C0326: Exactly one space required around assignment
scripts/kconfig/checkconfig.py:51:3: C0121: Comparison to None should be 'expr is None' (singleton-comparison)
scripts/kconfig/checkconfig.py:61:39: C0326: Exactly one space required after comma
scripts/kconfig/checkconfig.py:61:45: C0326: Exactly one space required after comma
scripts/kconfig/checkconfig.py:61:55: C0326: Exactly one space required after comma
scripts/kconfig/checkconfig.py:90:20: C0121: Comparison to True should be just 'expr' or 'expr is True' (singleton-comparison)
scripts/process_gperf.py:26:-1: W0105: String statement has no effect (pointless-string-statement)
scripts/sanity_chk/expr_parser.py:14:4: C0414: Import alias does not rename original package (useless-import-alias)
scripts/sanity_chk/expr_parser.py:15:4: C0414: Import alias does not rename original package (useless-import-alias)
scripts/sanity_chk/scl.py:54:23: W0613: Unused argument 'data' (unused-argument)
scripts/sanity_chk/scl.py:54:29: W0613: Unused argument 'schema' (unused-argument)
scripts/west_commands/build.py:172:15: C1801: Do not use `len(SEQUENCE)` to determine if a sequence is empty (len-as-condition)
scripts/west_commands/run_common.py:256:15: W0703: Catching too general exception Exception (broad-except)
scripts/west_commands/runners/core.py:236:8: R1703: The if statement can be replaced with 'var = bool(test)' (simplifiable-if-statement)
scripts/west_commands/runners/core.py:467:8: W0706: The except handler raises immediately (try-except-raise)
scripts/west_commands/runners/core.py:486:8: W0706: The except handler raises immediately (try-except-raise)
scripts/west_commands/runners/core.py:508:15: W1509: Using preexec_fn keyword which may be unsafe in the presence of threads (subprocess-popen-preexec-fn)
scripts/west_commands/runners/nrfjprog.py:56:11: C1801: Do not use `len(SEQUENCE)` to determine if a sequence is empty (len-as-condition)
scripts/west_commands/runners/qemu.py:13:4: W0235: Useless super delegation in method '__init__' (useless-super-delegation)
scripts/west_commands/runners/xtensa.py:15:4: W0235: Useless super delegation in method '__init__' (useless-super-delegation)
scripts/west_commands/sign.py:60:4: W0222: Signature differs from overridden '__call__' method (signature-differs)
```

ulfalizer, 2019-03-22:
CC @marc-hb 

Do you see any you think should be hard failures? Many of these are candidates for a "soft" check otherwise.

Think we should aim to keep any "soft" check non-spammy though. My experience is that people start ignoring stuff when there's too much of it.

marc-hb, 2019-03-22:
I don't have time to review them one by one right now, sorry.

From a quick look most of these should be run all the time because they're easy to fix and correct 99% of the time, however they should be filtered by `check_compliance.py class PylintSoftChecks` so the latter reports it only on the **modified files and lines** (exactly like checkpatch does) and doesn't bother the user with the 1% exceptions elsewhere in the file already granted 6 months ago.

> My experience is that people start ignoring stuff when there's too much of it.

100% correct.

nashif, 2021-05-28:
too nitty, too subjective, or too broad

closed 2021-05-28 by nashif