zephyrproject-rtos/zephyr #14807: disable SPIN_VALIDATE when SMP enabled

wentongwu, 2019-03-22:
**Describe the bug**
with both SPIN_VALIDATE and SMP(cpu_num = 2) on, when two different threads running on different cpu try to access the same one spinlock or semaphore, SPIN_VALIDATE will cause assert happen. So I suggest when SMP on, close SPIN_VALIDATE.

Another issue about SMP,  when do k_sem_give for a semaphore, I don't think it should do reschedule (z_reschedule in z_impl_k_sem_give) if there is no thread pending on that semaphore.

What have you tried to diagnose or workaround this issue?
disable SPIN_VALIDATE when SMP opened, assert will not happen.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=qemu_x86_64 ..
3. make
4. See error

**Expected behavior**
no assert happen.

andrewboie, 2019-03-25:
@andyross

wentongwu, 2019-03-26:
PR #14863 should fix this issue.

closed 2019-03-27 by wentongwu