zephyrproject-rtos/zephyr #14810: Custom driver implementation DT_<Driver Name>_LABEL undeclared

bokenator, 2019-03-22:
Hi,

I'm trying to implement a driver for the ADXL345 accelerometer based the ADXL372 driver which is already available on Zephyr. However, when I'm trying to use the driver in the application, the compilation fails with the following error:

```
../src/main.c: In function 'main':
../src/main.c:50:42: error: 'DT_ADI_ADXL345_0_LABEL' undeclared (first use in this function); did you mean 'DT_FLASH_AREA_0_LABEL'?
  struct device *dev = device_get_binding(DT_ADI_ADXL345_0_LABEL);
                                          ^~~~~~~~~~~~~~~~~~~~~~
                                          DT_FLASH_AREA_0_LABEL
../src/main.c:50:42: note: each undeclared identifier is reported only once for each function it appears in
../src/main.c:48:7: warning: unused variable 'meter' [-Wunused-variable]
```

I have already made edits to `CMakeLists.txt` and `Kconfig` in the `drivers/sensor` directory to add the new custom driver.

Any help would be greatly appreciated!
Thanks!

aescolar, 2019-03-22:
@galak : Could you help him?

bokenator, 2019-03-22:
Figured it out!

In addition to adding the driver to `CMakeLists.txt` and `Kconfig` in the `drivers/sensor` directory, we also need to create a `.yaml` file in `dts/bindings/sensor` directory.

closed 2019-03-22 by bokenator

thirtythreeforty, 2019-09-11:
@bokenator perhaps a bit off topic for this issue, but is your work-in-progress ADXL345 driver public anywhere?