zephyrproject-rtos/zephyr #14826: USB reset in suspended state

Qbicz, 2019-03-22:
**Describe the bug**
On startup, USB stack sends suspend and resume between CONNECTED and CONFIGURED states.

But sometimes USB stack will send RESET callback in between:

USB_DC_SUSPEND
USB_DC_RESET
USB_DC_RESUME

And we get RESUME when we are not suspended. Checked with @pawelzadrozniak - reset should not occur between suspend and resume.

**To Reproduce**
Using USB HID class. This sequence occurs right after boot, when boot is done with USB connected.

**Environment (please complete the following information):**
- nRF52840
- Zephyr 19de7ecc5b6dc1
```
CONFIG_USB=y
CONFIG_USB_NRF52840=y
CONFIG_USB_DEVICE_STACK=y
CONFIG_USB_DEVICE_HID=y
CONFIG_USB_DEVICE_LOG_LEVEL_OFF=y
CONFIG_USB_DRIVER_LOG_LEVEL_OFF=y
CONFIG_USB_HID_POLL_INTERVAL_MS=1
CONFIG_USB_HID_BOOT_PROTOCOL=y
CONFIG_USB_HID_PROTOCOL_CODE=2
```

Qbicz, 2019-03-22:
Full sequence of USB state changes I get is the same as in https://github.com/zephyrproject-rtos/zephyr/issues/14627

carlescufi, 2019-03-22:
@pawelzadrozniak will you be able to send a fix for this?

pawelzadrozniak, 2019-03-22:
> 
> 
> @pawelzadrozniak will you be able to send a fix for this?

That's something we have discussed earlier today.
I'll investigate next week what is causing the wrong event order.

Qbicz, 2019-04-17:
@pawelzadrozniak This issue no longer occurs on 06ab394f05c (v1.14.0)

The sequence after plugging a cable is now 
```
[04038208] <wrn> usb_state: USB connected
[04038310] <wrn> usb_state: USB suspend
[04042041] <wrn> usb_state: USB resume
[04042043] <wrn> usb_state: USB reset
[04045122] <wrn> usb_state: USB reset
[04050510] <wrn> usb_state: USB configured
```
There is no reset is suspended state.

closed 2019-04-17 by Qbicz