zephyrproject-rtos/zephyr #14827: cmake error

15658150157, 2019-03-22:
**Describe the bug**
I newly install ubuntu 18.04 on my pc, and follow the start guide, below is the error i meet.

 cmake -DBOARD=qemu_x86 ..
Zephyr version: 1.14.0
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.7", minimum required is "3.4") 
-- Selected BOARD qemu_x86
-- Loading /home/yhx/zephyrproject/zephyr/boards/x86/qemu_x86/qemu_x86.dts as base
-- Overlaying /home/yhx/zephyrproject/zephyr/dts/common/common.dts
/home/yhx/zephyrproject/zephyr/scripts/dts/extract_dts_includes.py:370: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  binding = merge_included_bindings(file, yaml.load(yf))
Traceback (most recent call last):
  File "/home/yhx/zephyrproject/zephyr/scripts/dts/extract_dts_includes.py", line 520, in <module>
    main()
  File "/home/yhx/zephyrproject/zephyr/scripts/dts/extract_dts_includes.py", line 503, in main
    load_bindings(root, args.yaml)
  File "/home/yhx/zephyrproject/zephyr/scripts/dts/extract_dts_includes.py", line 370, in load_bindings
    binding = merge_included_bindings(file, yaml.load(yf))
  File "/home/yhx/.local/lib/python3.6/site-packages/yaml/__init__.py", line 114, in load
    return loader.get_single_data()
  File "/home/yhx/.local/lib/python3.6/site-packages/yaml/constructor.py", line 43, in get_single_data
    return self.construct_document(node)
  File "/home/yhx/.local/lib/python3.6/site-packages/yaml/constructor.py", line 52, in construct_document
    for dummy in generator:
  File "/home/yhx/.local/lib/python3.6/site-packages/yaml/constructor.py", line 404, in construct_yaml_map
    value = self.construct_mapping(node)
  File "/home/yhx/.local/lib/python3.6/site-packages/yaml/constructor.py", line 210, in construct_mapping
    return super().construct_mapping(node, deep=deep)
  File "/home/yhx/.local/lib/python3.6/site-packages/yaml/constructor.py", line 135, in construct_mapping
    value = self.construct_object(value_node, deep=deep)
  File "/home/yhx/.local/lib/python3.6/site-packages/yaml/constructor.py", line 92, in construct_object
    data = constructor(self, node)
  File "/home/yhx/.local/lib/python3.6/site-packages/yaml/constructor.py", line 420, in construct_undefined
    node.start_mark)
yaml.constructor.ConstructorError: could not determine a constructor for the tag '!include'
  in "/home/yhx/zephyrproject/zephyr/dts/bindings/serial/ns16550.yaml", line 9, column 5
CMake Error at /home/yhx/zephyrproject/zephyr/cmake/dts.cmake:145 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  /home/yhx/zephyrproject/zephyr/cmake/app/boilerplate.cmake:396 (include)
  CMakeLists.txt:4 (include)

-- Configuring incomplete, errors occurred!

below is env:
env | grep ZE

ZEPHYR_TOOLCHAIN_VARIANT=zephyr
ZEPHYR_BASE=/home/yhx/zephyrproject/zephyr
ZEPHYR_SDK_INSTALL_DIR=/home/yhx/zephyr

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=qemu_x86 ..
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

aescolar, 2019-03-22:
@ulfalizer : Could you take a look?

carlescufi, 2019-03-22:
@15658150157 this is a known issue that was fixed already here: https://github.com/zephyrproject-rtos/zephyr/pull/14497. Please take the latest master

closed 2019-03-22 by carlescufi

ulfalizer, 2019-03-22:
@15658150157 
I think this is fixed by commit 4278bf5673 ("scripts/dts: Fix for PyYAML 5.1 Loader changes"), which you don't seem to have.

If you don't want to use the latest master, this workaround might work (which downgrades `pyyaml`):

    $ pip install --user pyyaml==3.13

CC @mike-scott

15658150157, 2019-03-23:
ok. solve it. very thanks.