zephyrproject-rtos/zephyr #14837: Bluetooth shell scan command parameter mandatory/optional evaluation is broken

jhedberg, 2019-03-22:
```
uart:~$ bt scan on
scan: wrong parameter count
scan - <value: on, passive, off> <dup filter: dups, nodups>
uart:~$ bt scan on nodups
Bluetooth active scan enabled
...
uart:~$ bt scan off
scan: wrong parameter count
scan - <value: on, passive, off> <dup filter: dups, nodups>
uart:~$ bt scan off dups
Scan successfully stopped
```

jarz-nordic, 2019-03-22:
@jhedberg : I have faced (and possibly caused) this problem.
This command shall accept:
`bt scan on` and `bt scan on dups/nodups` ?

jhedberg, 2019-03-22:
@jarz-nordic yes. The dups/nodups should be an optional parameter, whereas for "off" there should be no additional parameters. I suppose "on" and "off" need to be made into sub-commands so they can have different parameter requirements?

jarz-nordic, 2019-03-22:
I will fix that.
Haveing `on` and `off` subcommands would be great but this idea has been rejected some time ago ;-)

PR #14873 (2019-03-25): Bt shell fix accepted parameter count
Fixed accepted parameter count for command `bt scan`.

Fixes #14837
closed 2019-03-25 by jhedberg