zephyrproject-rtos/zephyr #14869: tests/lib/ringbuffer: Assertion failure at test_ring_buffer_main()

cinlyooi-intel, 2019-03-25:
**Describe the bug**
(memcmp((char *)getdata, rb_data, getsize * sizeof(u32_t)) == 0

Seen on Altera Max10

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=altera_max10
3. make
4. make run

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots or console output**
```
***** Booting Zephyr OS v1.14.0-rc1-1386-g5d35b6632765 (delayed boot 3000ms) *****
Running test suite test_ringbuffer_api
===================================================================
starting test - test_ringbuffer_init
PASS - test_ringbuffer_init
===================================================================
starting test - test_ringbuffer_declare_pow2
PASS - test_ringbuffer_declare_pow2
===================================================================
starting test - test_ringbuffer_declare_size
PASS - test_ringbuffer_declare_size
===================================================================
starting test - test_ringbuffer_put_get_thread
PASS - test_ringbuffer_put_get_thread
===================================================================
starting test - test_ringbuffer_put_get_isr
PASS - test_ringbuffer_put_get_isr
===================================================================
starting test - test_ringbuffer_put_get_thread_isr
PASS - test_ringbuffer_put_get_thread_isr
===================================================================
starting test - test_ringbuffer_pow2_put_get_thread_isr
PASS - test_ringbuffer_pow2_put_get_thread_isr
===================================================================
starting test - test_ringbuffer_size_put_get_thread_isr
PASS - test_ringbuffer_size_put_get_thread_isr
===================================================================
starting test - test_ring_buffer_main
    Assertion failed at /home/jenkins/workspace/zephyr-master-ci-master/zephyr.git/tests/lib/ringbuffer/src/main.c:95: test_ring_buffer_main: ((memcmp((char *)getdata, rb_data, getsize * sizeof(u32_t)) == 0) is false)**
data corrupted
FAIL - test_ring_buffer_main
===================================================================
starting test - test_ringbuffer_raw
PASS - test_ringbuffer_raw
===================================================================
starting test - test_ringbuffer_alloc_put
PASS - test_ringbuffer_alloc_put
===================================================================
starting test - test_byte_put_free
PASS - test_byte_put_free
===================================================================
Test suite test_ringbuffer_api failed.
===================================================================
```

**Environment (please complete the following information):**
 - OS: fedora29
 - Toolchain: zephyr
 - Commit SHA or Version used: 5d35b6632765bdb4f9deae432bfb378f64f85fa3

**Additional context**
Add any other context about the problem here.

andrewboie, 2019-03-27:
does not fail on qemu_nios2

andrewboie, 2019-04-09:
I think this has something to do with unaligned memory access to the ring buffer storage area, still digging

andrewboie, 2019-04-09:
Seems that on the real Max10 board, trying to array reference an unaligned pointer implictly changes stuff. The rb_data[] array passed to ring_buf_item_put() is not 4-byte aligned.

In emulation, and presumably on other hardware, we see:

```
RB PUT TO ADDR 0x00400e47
data[i] is 44434241
```

But on the Max10 board, the unaligned pointer is treated differently:

```
RB PUT TO ADDR 0x00400e33
data[i] is 41000000
```

We expect data[i] to be 0x44434241, which is what "ABCD" would look like if cast to a u32_t. It seems that the CPU is instead doing something which causes the array beginning to be 0x00400e30 instead.

andrewboie, 2019-04-09:
https://stackoverflow.com/questions/51126257/what-does-the-standard-say-about-unaligned-memory-access

I'm gonna go with this being undefined behavior.

closed 2019-04-09 by nashif

arun1joshi, 2019-04-10:
verified that the tests/lib/ringbuffer test cases are passed with the latest commit 9926f94

PR #15311 (2019-04-09): tests: ringbuffer: avoid unaligned mem access
Casting the rb_data character array to a u32_t can result
in an unaligned u32_t * pointer being passed to
ring_buf_item_put(), since rb_data is only byte-aligned.

Our Altera Max10 CPU build is not configured to detect
unaligned memory access and throw an exception, it is
instead rounding down the memory address
of the data pointer to the nearest 4-byte value, causing
the wrong data to be copied into the ring buffer.

It appears that in the C standard this is considered
Undefined Behavior so the approach this patch takes is
to fix the test, by ensu [... cut]
commit 9926f947 (2019-04-10): tests: ringbuffer: avoid unaligned mem access

Casting the rb_data character array to a u32_t can result
in an unaligned u32_t * pointer being passed to
ring_buf_item_put(), since rb_data is only byte-aligned.

Our Altera Max10 CPU build is not configured to detect
unaligned memory access and throw an exception, it is
instead rounding down the memory address
of the data pointer to the nearest 4-byte value, causing
the wrong data to be copied into the ring buffer.

It appears that in the C standard this is considered
Undefined Behavior so the approach this patch takes is
to fix the test, by ens [... cut]