zephyrproject-rtos/zephyr #14870: samples/mpu/mpu_stack_guard_test: Fatal fault in ISR! Spinning

cinlyooi-intel, 2019-03-25:
**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=frdm_k64f
3. make
4. make run

**Screenshots or console output**
```
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-1386-g5d35b6632765 (delayed boot 1000ms) *****
STACK_ALIGN 0x8
MPU STACK GUARD Test
Canary Initial Value = 0xf0cacc1a threads 0x20000ea0
Canary = 0x0000000f	Test not passed.
***** BUS FAULT *****
  Instruction bus error
  NXP MPU error, port 3
    Mode: Supervisor, Instruction Address: 0x20000ed0
    Type: Read, Master: 0, Regions: 0x8800
***** Hardware exception *****
Current thread ID = 0x20000ea0
Faulting instruction address = 0x20000ed0
Fatal fault in thread 0x20000ea0! Aborting.
***** HARD FAULT *****
  Fault escalation (see below)
***** USAGE FAULT *****
  Illegal use of the EPSR
***** Hardware exception *****
Current thread ID = 0x20000ea0
Faulting instruction address = 0x0
Fatal fault in ISR! Spinning...
```
**Environment (please complete the following information):**
 - OS: fedora29
 - Toolchain: zephyr
 - Commit SHA or Version used: 5d35b6632765bdb4f9deae432bfb378f64f85fa3

ioannisg, 2019-03-25:
I think this is expected. Try to execute with CONFIG_HW_STACK_PROTECTION=y (I think it is set in the alternative prj.conf). You should expect _not_ to see: "Test not passed".

agansari, 2019-03-25:
@cinlyooi-intel building with:

> cmake -DBOARD=frdm_k64f -DCONF_FILE=prj_stack_guard.conf ..

Will make the test pass as mentioned above. 
I can't reproduce ***** HARD FAULT ***** part.

cinlyooi-intel, 2019-03-26:
@ioannisg : You are right. Examining sample.yaml says we are expecting the console output I am seeing. 
@agansari : With CONF_FILE=prj_stack_guard.conf (sample.yaml's test_stack_guard testcase ) I don't see the `Hard Fault` either. `Hard Fault` only with "sample.yaml's test testcase"

I am closing this test case.

cinlyooi-intel, 2019-03-26:
I am reopenning this because I noticed the "Test not passed"  mentioned by @ianisg on the same line as "Canary = 0x0000000f". Missed that initially

PR #14948 (2019-04-02): samples: remove mpu_stack_guard_test
This old "test" has been superseded by test cases in
tests/kernel/mem_protect.

Fixes: #14870
commit 9a8f7451 (2019-04-02): samples: remove mpu_stack_guard_test

This old "test" has been superseded by test cases in
tests/kernel/mem_protect.

Fixes: #14870
closed 2019-04-02 by nashif