zephyrproject-rtos/zephyr #14871: tests/posix/fs : Deleted

cinlyooi-intel, 2019-03-25:
**Describe the bug**
When running the test case, I get
```
--- 40 messages dropped ---
```

Please note that the number in the message changes. In general, the ztest test case one is trying to run the larger the number.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=nrf52840_pca10056
3. make
4. See error

**Impact**
 - The test harness in testcase.yaml will report that the test case failed. It is relying on finding all the expected `Pass - test_fs_xxx` lines but cannot get all of them, because a significant number of them are in part of the messages that were deleted.
 - Visually, we cannot confirm all test cases passed because we are not seeing all the `Pass - test_fs_xxx` lines. Instead, we trust `Test  suite posix_fs_test succeeded` to confirm to us that all test passed.
 
**Screenshots or console output**
```
Created dir /RAM:/testdir!
--- 40 messages dropped ---
PASS - test_fs_mkdir
===================================================================
starting test - test_fs_readdir

readdir test:

Listing dir /RAM:/testdir:
[FILE] TESTFILE.TX
PASS - test_fs_readdir
===================================================================
Test suite posix_fs_test succeeded
===================================================================
```

**Environment (please complete the following information):**
 - OS: fedora29
 - Toolchain : zephyr
 - Commit SHA or Version used: 5d35b6632765bdb4f9deae432bfb378f64f85fa3

carlescufi, 2019-03-26:
@cinlyooi-intel this is just a result of a buffer that is too small. You can increase the buffer size with:
```
CONFIG_LOG_BUFFER_SIZE=4096
```
in your `prj.conf`.

Since there are memory constraints in some of the boards I don't want to increase that size for all boards.

closed 2019-03-26 by carlescufi

aescolar, 2019-03-26:
Duplicate: #14903