zephyrproject-rtos/zephyr #14898: samples/subsys/nvs - error

chiachuanwu, 2019-03-26:
Steps to reproduce the behavior:

mkdir build; cd build
cmake -DBOARD=nrf51_pca10028
make
make run

Screenshots or console output
```
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-1417-ge2de07b2f151 (delayed boot 1000ms) *****
```

Environment (please complete the following information):

OS: fedora29
Toolchain: zephyr
Commit SHA or Version used: e2de07b2

Laczen, 2019-03-26:
Hi @chiachuanwu, that is strange, I just tried to run the sample and it works just fine. Why are you using make run ? The correct commands to build and run the sample are:
```
mkdir build; cd build
cmake -DBOARD=nrf51_pca10028 ..
make
make flash
```

carlescufi, 2019-03-26:
@chiachuanwu in your report you write:

> make run

Are you flashing the board with `make flash`?

chiachuanwu, 2019-03-29:
hi @carlescufi  @Laczen , sorry for late reply, i had updated the reproduced steps.
The target board should be "nrf52840_pca10056 ", instead of "nrf51_pca10028 " . 

I tested this sample on "nrf51_pca10028 " target board, the console output log looks correct.

But on "nrf52840_pca10056 " target board, the console output log looks not the same:
```
Rebooting in ...561 --- 114 messages dropped ---
62 63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f 70 71 72 73 74 75 76 77 78 79 7a 7b 7c 7d 7e 7f
...348--- 449 messages dropped ---
...347...346...345...344...343...342...341...340...339...338...337...336...335...334...333...332...331...330...329...328...327...326...325...324...323...323
Oldest reboot counter: 323
Rebooting in ...561 --- 114 messages dropped ---
62 63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f 70 71 72 73 74 75 76 77 78 79 7a 7b 7c 7d 7e 7f
...348--- 450 messages dropped ---
...347...346...345...344...343...342...341...340...339...338...337...336...335...334...333...332...331...330...329...328...327...326...325...324...323...323
Oldest reboot counter: 323
Rebooting in ...561 --- 114 messages dropped ---
62 63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f 70 71 72 73 74 75 76 77 78 79 7a 7b 7c 7d 7e 7f
...348--- 451 messages dropped ---
...347...346...345...344...343...342...341...340...339...338...337...336...335...334...333...332...331...330...329...328...327...326...325...324...323...323
Oldest reboot counter: 323
Rebooting in ...561 --- 114 messages dropped ---
62 63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f 70 71 72 73 74 75 76 77 78 79 7a 7b 7c 7d 7e 7f
...348--- 452 messages dropped ---
...347...346...345...344...343...342...341...340...339...338...337...336...335...334...333...332...331...330...329...328...327...326...325...324...323...323
Oldest reboot counter: 323
Rebooting in ...561 --- 114 messages dropped ---
62 63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f 70 71 72 73 74 75 76 77 78 79 7a 7b 7c 7d 7e 7f
...348--- 453 messages dropped ---
...347...346...345...344...343...342...341...340...339...338...337...336...335...334...333...332...331...330...329...328...327...326...325...324...323...323
Oldest reboot counter: 323
Rebooting in ...561 --- 114 messages dropped ---
62 63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f 70 71 72 73 74 75 76 77 78 79 7a 7b 7c 7d 7e 7f
Reboot counter reached max value.
Reset to 0 and exit test.
```

Laczen, 2019-03-29:
Hi @chiachuanwu, the sample seems to run perfectly. The messages you are seeing are related to the logger, you have quite a lot of dropped messages. If you consider these messages as a bug you should file a bug for the logger: "Logger: to many dropped messages when running sample samples/subsys/nvs".

chiachuanwu, 2019-03-29:
hi @Laczen noted, i will close this issue.

closed 2019-03-29 by chiachuanwu