zephyrproject-rtos/zephyr #14899: [Bluetooth controller] ACL data packets stall

mkarhumaa, 2019-03-26:
**My test setup** 
nRF52840_pca10059, hci_usb sample. Proprietary host stack and application (fuzzer) in Linux
Peer: csr8510, Linux kernel

Proprietary application is used for Bluetooth SMP fuzzing. Test target is running also linux.

**Describe the bug**
After certain test cases Zephyr controller goes to the state that it does not send HCI ACL packets out. Control packets are still working fine: i.e. connection can be created. When controller is in bad state, host stack sends HCI ACL packets to controller until controller tx buffer is full but host never receives Number of completed packets events. Dongle needs to be unplugged and plugged again to recover.

Configuring `CONFIG_BT_CTLR_TX_BUFFERS=1` makes this problem not reproducible.

Hci logs and sniffer captures as attachment: [zephyr_captures.zip](https://github.com/zephyrproject-rtos/zephyr/files/3006734/zephyr_captures.zip)

**To Reproduce**
I can reproduce this reliably. Requires proprietary application. 

Test sequence consists of sending 1024 bytes of 0xff, valid public key, pairing random, dhkey check and trying to start encryption. This is done many times until ACL packets are stalling.

**Expected behavior**
Controller is expected to send ACL packets out normally.

**Impact**
Stopper.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK
 - Commit 972221423cde92639f685542b10a8caa35793ec5

mkarhumaa, 2019-03-26:
Thank you @cvinayak for looking into this!

I will try with hci_uart example once I've got dev kit. I guess I cannot test it with nrf52840_pca10059?

Still there is a thing that really bothers me: I consistently see that after sending Start encryption request to controller, if host receives Number of completed packets only three times before getting Encryption change event, on next connection ACL data will stall.

See two screenshots below.

First one is screenshot from case with no problems, ACL data does not stall on next connection.
[image]

Second one is the case, where on next connection ACL data will stall.
[image]

This behaviour is consistent but I'm unsure whether there is correlation.

carlescufi, 2019-03-26:
@mkarhumaa 

>I will try with hci_uart example once I've got dev kit. I guess I cannot test it with nrf52840_pca10059?

Not unless you have an external debugger.

I am very interested in this behaviour. I know we already discussed this on Slack, but can you still double-check whether disabling `CONFIG_BT_HCI_ACL_FLOW_CONTROL` has any effect?
The other option is what @cvinayak mentions, the fact that USB uses different endpoints for commands and data, which leads to unsolvable race conditions. @jhedberg might be able to comment on this as well.

mkarhumaa, 2019-03-27:
@carlescufi , thanks for your comments!

> I am very interested in this behaviour. I know we already discussed this on Slack, but can you still double-check whether disabling CONFIG_BT_HCI_ACL_FLOW_CONTROL has any effect?
I re-tested with config `CONFIG_BT_HCI_ACL_FLOW_CONTROL=n` (confirmed from build/zephyr/.config) and I don't see any difference. ACL packet TX still stalls after a while. I don't actually know what difference I should see in hci log when `CONFIG_BT_HCI_ACL_FLOW_CONTROL` is disabled. To my understanding it should affect received ACL data?

> The other option is what @cvinayak mentions, the fact that USB uses different endpoints for commands and data, which leads to unsolvable race conditions.
I verified that once controller is in bad state and TX stalls, ACL data RX still works fine. Are the ACL RX and TX using same USB endpoint?

I used following setup:
nrf52840_pca10059 and Zephyr hci_usb, Linux bluetoothctl
Peer: Android 8 phone, nRF connect as a gatt server.

First I used our application to get controller into bad state, then changed to bluetoothctl (hci reset was done), scanned, connected to android device. nRF connect tried to do Gatt service discovery but did not get any response so it disconnected. I have hci logs from both sides and it clearly shows that on zephyr side RX works but TX does not.

mkarhumaa, 2019-03-27:
Ok, thinks are getting clearer now. It indeed seems that problem is somewhere in USB endpoint.

I captured both usb and hci traffic to same pcap. From the pcap we can see that once things start go wrong we do not get `URB_BULK out`s with type `URB_COMPLETE` for those `URB_BULK out`s with type `URB_SUBMIT` we have sent. Once we give up and disconnect we finally get ÙRB_BULK out` with type `URB_COMPLETE` and status `-ENOENT` whereas the status normally is `Success`.

Capture is as attachment: [tx-stall-usb-bt-capture.pcapng.zip](https://github.com/zephyrproject-rtos/zephyr/files/3012492/tx-stall-usb-bt-capture.pcapng.zip)

Any hints where should I look next to debug this further?

mkarhumaa, 2019-03-27:
After all this ended up being problem in `subsystem/usb/class/bluetooth.c`.

If we ran out of buffers in acl_tx_pool we stopped reading stuff from hci tx endpoint.

See pr: https://github.com/zephyrproject-rtos/zephyr/pull/14936

PR #14936 (2019-03-28): usb: Fix ACL data tx stalling
This patch increases size of acl_tx_pool and makes sure that acl_read_cb
waits for available buffers.

Previously it was possible to run out of buffers and in that case
acl_read_cb returned. This caused ACL data TX to stall because device
did not read data from HCI out endpoint.

Fixes #14899
commit 933acfef (2019-03-28): usb: Fix ACL data tx stalling

This patch increases size of acl_tx_pool and makes sure that acl_read_cb
waits for available buffers.

Size of acl_tx_pool is now configurable with CONFIG_BT_CTLR_TX_BUFFERS.

Previously it was possible to run out of buffers and in that case
acl_read_cb returned. This caused ACL data TX to stall because device
did not read data from HCI out endpoint.

Fixes #14899
closed 2019-03-28 by carlescufi