zephyrproject-rtos/zephyr #14904: Sanitycheck report not clear if built or ran a test

PiotrZierhoffer, 2019-03-26:
When running sanitycheck for a platform that supports emulation, you don't really know if the test is just built or ran in the emulator.

For example, to test `m2gl025` board with support in Renode, running the following command:
`./scripts/sanitycheck  --platform m2gl025_miv -v`

gets you 124 `PASSED` tests regardless of having Renode installed or not.

This makes perfect sense that these tests pass, but it's not clear if they were actually started or not. Having this indication would help verifying if the system setup is correct to perform the tests.

A clear output that the test passed BUT was only built would help a lot. A green `PASSED - BUILD ONLY` or just `BUILT` output should be clear enough.

I don't know if this will not break other scripting in sanitycheck-related services, but if not it should be fairly easy to implement, provided we agreed on the format.

marc-hb, 2019-04-22:
fixed by PR #15523 ?

marc-hb, 2019-04-30:
I think it's fixed by PR #15523, re-open if I misunderstood.

closed 2019-04-30 by marc-hb