zephyrproject-rtos/zephyr #14906: USB: NXP Device controller does not pass testusb tests

finikorg, 2019-03-26:
We have testusb sample for testing USB device controller, at the moment frdm_k64f does not pass all tests.
https://docs.zephyrproject.org/latest/samples/subsys/usb/testusb/README.html

jfischer-phytec-iot, 2019-03-26:
test 10 --> 71 (Protocol error)
[1288506.850622] usbtest 2-5:1.0: TEST 10:  queue 32 control calls, 1000 times
[1288507.095285] usbtest 2-5:1.0: subtest 5 error, status -71
[1288507.095296] usbtest 2-5:1.0: control queue 80.06, err -71, 29322 left, subcase 5, len 0/10
[1288507.095354] usbtest 2-5:1.0: subcase 6 completed out of order, last 4
[1288507.095613] usbtest 2-5:1.0: subcase 5 completed out of order, last 5

?

jfischer-phytec-iot, 2019-03-26:
I tested it for a while on different hosts and host controllers. It works fine on my Thinkpad (Intel 8 Series USB xHCI and EHCI), all the tests 0..29 passes. I noticed sometimes it failed during test 9 or test 10 on other host, but then passes with reduced number of iterations e.g. `./testusb -c 100 -t 10 -D /dev/bus/usb/002/006`.

agansari, 2019-08-07:
@finikorg @jfischer-phytec-iot can you redo this test? I think it has the same cause as #17007 which was fixed.

jfischer-phytec-iot, 2019-08-21:
Currently no one platform passes the test, I will keep it open bur remove NXP label.

jfischer-phytec-iot, 2019-09-16:
Fixed now, among other by:
60c4882badb895d37cc0f43fa34483f847b407f3
9ba269f580bb754ef566129eab262106c3afacac
28e97c7f3a4d63546053abc2f65affe36046124d

closed 2019-09-16 by jfischer-phytec-iot