zephyrproject-rtos/zephyr #14907: Can't seem to use RST formatting in .h file (i.e. doxygen?) docstrings

pfalcon, 2019-03-26:
I'm trying to put following as a docstring in .h file:
~~~
/**
 * @brief Create a network socket
 *
 * @details
 * See `POSIX.1-2017 article <http://pubs.opengroup.org/onlinepubs/9699919799/functions/socket.html>`_
 * for normative description. This function is also exposed as ``socket()``
 * if :option:`CONFIG_NET_SOCKETS_POSIX_NAMES` is defined.
 */
__syscall int zsock_socket(int family, int type, int proto);
~~~

But the generated sphinx HTML is badly formatted and there's no really links for that URL/option. @dbkinder, please suggest how to deal with these.

[image]

dbkinder, 2019-03-26:
You're right that embedding reST formatting in doxygen comments might render as you'd expect.  To fix this you need to explicitly tell doxygen that the enclosed lines are reST with @rststar and @endrststar markers:

```
 * @details
 * @rststar 
 * See `POSIX.1-2017 article <http://pubs.opengroup.org/onlinepubs/9699919799/functions/socket.html>`_
 * for normative description. This function is also exposed as ``socket()``
 * if :option:`CONFIG_NET_SOCKETS_POSIX_NAMES` is defined.
 * @endrststar
 */
```
The @rststar doxygen directive is an ALIAS (defined in doc/zephyr.doxyfile.in) that tells doxygen to create a verbatim block that contains reST formatted text and the leading asterisk on each line should be removed.

Put doxygen comments that have reST formatting within these markers and it should be rendered as you'd expect when it comes out the other side.  We normally don't need to explicitly use these markers because most doxygen formatting constructs are handled well, but the more interesting reST/Sphinx constructs aren't.  For those, we need to let doxygen/Sphinx know what's up.

I'll add mention of this to the [documentation contributing guide](https://docs.zephyrproject.org/latest/guides/documentation/index.html).

pfalcon, 2019-03-26:
Thanks for the quick response, works like a charm! And thanks for verbatim example, if you'd just mentioned `@rststar`, I'd probably have written it as `@rststart` and then wondered why it doesn't work ;-).

closed 2019-03-26 by pfalcon