zephyrproject-rtos/zephyr #14918: olimexino stm32 CAN errors

SebastianStumm, 2019-03-26:
**CAN Support Olimexino STM32**

Hello,
First I'm new to Zephyr so there is a high chance that I missed something.
As far as i understand the documentation on the Olimexino stm32 there is no CAN support yet since it's not listed in supported features?

> Other hardware features are not supported by the Zephyr kernel.

I tried to get the CAN sample running with the CAN Driver (1.14.0-rc2) but I just can't figure out if I'm doing something wrong in the configuration or if it's just not supported.

My proj.conf has the following settings:
>CONFIG_CAN=y
>CONFIG_CAN_STM32=y
>CONFIG_CAN_INIT_PRIORITY=80
>CONFIG_CAN_1=y
>CONFIG_CAN_MAX_FILTER=5

I use cmake like: 

> cmake -GNinja -DBOARD=olimexino_stm32 ..

If I'm trying to compile the sample CAN code with ninja i get many errors like the following:

> drivers/can/stm32_can.c:855:24: error: 'DT_CAN_1_BASE_ADDRESS' undeclared here (not in a function); did you mean 'DT_SPI_1_BASE_ADDRESS'?

I figured out that the olimexino_stm32 uses the stm32f103XX soc which has no CAN configuration at all compared to the stm32F03XX in the "dts_fixup.h" file. Also in the "pinmux.c" file of the olimexino_stm32 board there are no pins configured for CAN-RX/TX.

My Question is if there is already CAN support for the olimexino_stm32 before I continue trying to get it to work?

carlescufi, 2019-03-26:
@SebastianStumm `DT_CAN_1_BASE_ADDRESS` is the macro required by CAN that is supposed to be generated by Device Tree. If this board doesn't have DT nodes defined for CAN then they need to be added.

alexanderwachter, 2019-03-26:
Welcome to Zephyr @SebastianStumm !
We didn't add CAN support for the olimexino board so far. You can add the CAN controller to the dts for the stm32f103 series and configure the pins (PB8 as CANrx and PB9 as CANtx) in pinmux of the olimexino board. Basically the stm32 CAN driver should work for the stm32f1 series.
Don't forget to turn off the loopback mode when connecting two boards. Otherwise you won't see any messages on the bus.
If it works, pleas submit a PR so we can add the board to the list of supported CAN boards.

alexanderwachter, 2019-04-05:
Hy @SebastianStumm!
How are you doing with the CAN on the olimexino?
I just created a PR for another board (#15209). You can see there how to bring up CAN on a board.

SebastianStumm, 2019-04-09:
Hi,
First of all thank you alot for the help and feedback !
I did add the controller but did not submit a pr yet because it does not fully work currently. I took a look at your example with the nucleo_f746zg and I basicly already did the same configuration with some minor changes due the lack of alternate function registers of the stm321f03.

**1. stm32f103Xb.dtsi under soc**
```
can1: can@40006400 {
	compatible = "st,stm32-can";
	#address-cells = <1>;
	#size-cells = <0>;
	reg = <0x40006400 0x400>;
	interrupts = <19 0>, <20 0>, <21 0>, <22 0>;
	interrupt-names = "TX", "RX0", "RX1", "SCE";
	clocks = <&rcc STM32_CLOCK_BUS_APB1 0x02000000>;
	status = "disabled";
	label = "CAN_1";
};
```

**2. olimexino_stm32.dts**
```
&can1 {
	bus-speed = <125000>;
	sjw = <1>;
	prop_seg_phase_seg1 = <6>;
	phase_seg2 = <5>;
	status = "ok";
};
```

**3. pinmux.c (in the olimex_stm32 folder)**
```
#ifdef CONFIG_CAN_1
	{STM32_PIN_PB8, STM32F1_PINMUX_FUNC_PB8_CAN_RX},
	{STM32_PIN_PB9, STM32F1_PINMUX_FUNC_PB9_CAN_TX},
#endif /* CONFIG_CAN_1 */
```

**4. pinxmus_stm32f1.h**
```
#define STM32F1_PINMUX_FUNC_PB9_CAN_TX			 STM32_PIN_CAN_TX
#define STM32F1_PINMUX_FUNC_PB8_CAN_RX 			 STM32_PIN_CAN_RX
```

**5. stm32-pinctrlf1.h**
```
#define STM32_PIN_CAN_TX        (STM32_MODE_OUTPUT | STM32_CNF_ALT_FUNC | STM32_CNF_PUSH_PULL | STM32_PUPD_PULL_UP)
#define STM32_PIN_CAN_RX        (STM32_MODE_INPUT | STM32_CNF_IN_FLOAT)
```

**6. dts-fixup.h**
```
#define DT_CAN_1_BASE_ADDRESS		DT_ST_STM32_CAN_40006400_BASE_ADDRESS
#define DT_CAN_1_BUS_SPEED		DT_ST_STM32_CAN_40006400_BUS_SPEED
#define DT_CAN_1_NAME			DT_ST_STM32_CAN_40006400_LABEL
#define DT_CAN_1_IRQ_TX			DT_ST_STM32_CAN_40006400_IRQ_TX
#define DT_CAN_1_IRQ_RX0		DT_ST_STM32_CAN_40006400_IRQ_RX0
#define DT_CAN_1_IRQ_RX1		DT_ST_STM32_CAN_40006400_IRQ_RX1
#define DT_CAN_1_IRQ_SCE		DT_ST_STM32_CAN_40006400_IRQ_SCE
#define DT_CAN_1_IRQ_PRIORITY		DT_ST_STM32_CAN_40006400_IRQ_0_PRIORITY
#define DT_CAN_1_SJW			DT_ST_STM32_CAN_40006400_SJW
#define DT_CAN_1_PROP_SEG_PHASE_SEG1	DT_ST_STM32_CAN_40006400_PROP_SEG_PHASE_SEG1
#define DT_CAN_1_PHASE_SEG2		DT_ST_STM32_CAN_40006400_PHASE_SEG2
#define DT_CAN_1_CLOCK_BUS		DT_ST_STM32_CAN_40006400_CLOCK_BUS
#define DT_CAN_1_CLOCK_BITS		DT_ST_STM32_CAN_40006400_CLOCK_BITS
```

The CAN sample compiles and flashes now without any errors. 
But the sample behaves not as expected.
When reseting the board I get the sample loaded without any problems and the output (from putty) looks like this:

> ▒***** Booting Zephyr OS v1.14.0-rc3-128-ged139948a54b *****
Finished init. waiting for Interrupts
TX thread is running.
filter id: 1

But when I press the user button the sample does not behave as expected. Occasionally some messages are transmitted but they don't correspond to the amount of button clicks. Sometimes there are 8-10 messages from a single button click, sometimes a single message from multiple button clicks and most of the times no messages at all.

So by now I dug into the stm32f103xx reference manual and found a section under the General-purpose and alternate-function I/Os (GPIOs and AFIOs) registers that states that CAN is by default not mapped to PB8&PB9 but PA11&PA12. So without any change PB8&PB9 are by default set to TIM4_CH3&TIM4_CH4.
I do not know if the unexpected behaviour is rooted in me misconfiguring the CAN controller or if I need to remap the pins first. And if so how do I achieve this within zephyr? I found the following in stm321fxx_ll_gpio.h:
```
#if defined(AFIO_MAPR_CAN_REMAP_REMAP1)

/**
  * @brief Enable or disable the remapping of CAN alternate function CAN_RX and CAN_TX in devices with a single CAN interface.
  * @rmtoll MAPR          CAN_REMAP           LL_GPIO_AF_RemapPartial1_CAN1
  * @note  CASE 1: CAN_RX mapped to PA11, CAN_TX mapped to PA12
  * @retval None
  */
__STATIC_INLINE void LL_GPIO_AF_RemapPartial1_CAN1(void)
{
  CLEAR_BIT(AFIO->MAPR, AFIO_MAPR_CAN_REMAP);
  SET_BIT(AFIO->MAPR, AFIO_MAPR_CAN_REMAP_REMAP1);
}

/**
  * @brief Enable or disable the remapping of CAN alternate function CAN_RX and CAN_TX in devices with a single CAN interface.
  * @rmtoll MAPR          CAN_REMAP           LL_GPIO_AF_RemapPartial2_CAN1
  * @note  CASE 2: CAN_RX mapped to PB8,  CAN_TX mapped to PB9 (not available on 36-pin package)
  * @retval None
  */
__STATIC_INLINE void LL_GPIO_AF_RemapPartial2_CAN1(void)
{
  CLEAR_BIT(AFIO->MAPR, AFIO_MAPR_CAN_REMAP);
  SET_BIT(AFIO->MAPR, AFIO_MAPR_CAN_REMAP_REMAP2);
}

/**
  * @brief Enable or disable the remapping of CAN alternate function CAN_RX and CAN_TX in devices with a single CAN interface.
  * @rmtoll MAPR          CAN_REMAP           LL_GPIO_AF_RemapPartial3_CAN1
  * @note  CASE 3: CAN_RX mapped to PD0,  CAN_TX mapped to PD1
  * @retval None
  */
__STATIC_INLINE void LL_GPIO_AF_RemapPartial3_CAN1(void)
{
  CLEAR_BIT(AFIO->MAPR, AFIO_MAPR_CAN_REMAP);
  SET_BIT(AFIO->MAPR, AFIO_MAPR_CAN_REMAP_REMAP3);
}
#endif
```
and similar code in the sm32f1xx_hal_gpio_ex.h

From this I see that there is a possibility to remap the CAN alternate function but I could not figure out where and how to call the remap function. Therefor i could not test if this causes the problem. I also noticed that all the jumpers on the board are smd and therefor the CAN-Bus (CAN_T JMPR) is not terminated but I guess this should not cause any problems in loopback mode?
So my questions are first of all if the configuration of the CAN controller is correct and if so how to remap in zephyr? 

Thanks in advance & best regards
Sebastian

alexanderwachter, 2019-04-09:
@SebastianStumm can you try calling
```C
LL_GPIO_AF_RemapPartial2_CAN1();
```
in zephyr/boards/arm/olimexino_stm32/pinmux.c after
```C
stm32_setup_pins(pinconf, ARRAY_SIZE(pinconf));
```

The phenomenon of sending more than one frame could also be a result of bouncing user button.
Enable Asserts and debug log of the CAN driver in menuconfig. Maybe something is wrong with the BS1 and BS2 config, for example the prescaler is not a real number.

SebastianStumm, 2019-04-10:
If I try to call the remap function it's undeclared.
> undefined reference to `LL_GPIO_AF_RemapPartial2_CAN1'

If I inlcude the the header file it does compile
```
#include <stm32f1xx_ll_gpio.h>
```
It did not change anything in the behaviour of the sample.

I enabled debug log for CAN, could not find a way to turn on assertion for CAN specificly just globally.
```
CONFIG_LOG=y
CONFIG_CAN_LOG_LEVEL_DBG=y
CONFIG_ASSERT=y
```

My output before pressing the user button now is:
```
▒***** Booting Zephyr OS v1.14.0-rc3-128-ged139948a54b *****
Finished init. waiting for Interrupts
TX thread is running.
filter id: 1
[00:00:00.000,000] <dbg> stm32_can.can_stm32_runtime_configure: Runtime configure of CAN_1 done
[00:00:00.000,000] <dbg> stm32_can.config_can_1_irq: Enable CAN1 IRQ
[00:00:00.000,000] <inf> stm32_can: Init of CAN_1 done
[00:00:00.000,000] <dbg> stm32_can.can_stm32_runtime_configure: Runtime configure of CAN_1 done
[00:00:00.000,000] <dbg> stm32_can.can_stm32_set_filter: Setting filter ID: 0x1, mask: 0x7ff
[00:00:00.000,000] <dbg> stm32_can.can_stm32_set_filter: Filter type: standard ID without mask (1)
[00:00:00.000,000] <dbg> stm32_can.can_stm32_set_filter: Filter set! Filter number: 0 (index 0)
[00:00:00.010,000] <dbg> stm32_can.can_stm32_set_filter: Setting filter ID: 0x10, mask: 0x7ff
[00:00:00.010,000] <dbg> stm32_can.can_stm32_set_filter: Filter type: standard ID without mask (1)
[00:00:00.010,000] <dbg> stm32_can.can_stm32_set_filter: Filter set! Filter number: 1 (index 1)
[00:00:00.010,000] <dbg> stm32_can.can_stm32_set_filter: Setting filter ID: 0x12345, mask: 0x1fffffff
[00:00:00.010,000] <dbg> stm32_can.can_stm32_set_filter: Filter type: extended ID without mask (3)
[00:00:00.010,000] <dbg> stm32_can.can_stm32_set_filter: Filter set! Filter number: 4 (index 4)
```

After I press the button it does drop a lot of messages for just pressing the button once or twice. Also it does not stop and randomly drops messages and "presses" the button by itself without any interaction from my part.
```
Button pressed! Send message 1
Button pressed 1 times
Button pressed! Send message 0
[00:01Button pressed 2 times
:11.83String sent over CAN
0,000] <dbg> stm32_can.can_stm32_send: Sending 1 bytes on CAN_1. Id: 0x10, ID type: standard, Remote Frame: no
[00:01:11.830,000] utton pressed! Send message 1
[0m<dbButton pressed 3 times
g> stm32_can.can_stm32_send: Using mailbox 0
--- 1 messages dropped ---
[00:01:11.830,000] <dbg> stm32_can.can_stm32_send: Using mailbox 1
[00:0Button pressed! Send message 0
1:11.8Button pressed 4 times
30,00String sent over CAN
0] <dbg> stm32_can.can_stm32_rx_isr_handler: Message on filter index 1
--- 13 messages dropped ---
[00:01:11.850,000]Button pressed! Send message 1
 <Button pressed 5 times
dbg> stm32_can.can_stm32_rx_isr_handler: Message on filter index 4
--- 5 messages dropped ---
[00:01:11.870,000] <dbg> stm32_can.can_stm32_send: Using Button pressed! Send message 0
mailboButton pressed 6 times
x 1
   tring sent over CAN
0m
--- 14 messages dropped ---
Button pressed! Send message 1_can.can_stm32_rx_isr_handler: Message on filter index 4

utton pressed 7 times
1m--- 4 messages dropped ---
[00:01:11.910,000] <dbg> stm32_can.can_stm32_send: Using mailbox 1
[... 68 lines cut ...]
ropped ---
[00:01:12.190,000] <dbg> stm32_can.can_stm32_send: Using mailbox 1
--- 6 messages dropped ---
[00:01:12.190,000] <dbg> stm32_can.can_stmButton pressed! Send message 0
32_rx_Button pressed 22 times
isr_hString sent over CAN
andler: Message on filter index 1
--- 13 messages dropped ---
[00:01:12.210,000] <dbg> stm32_can.can_stm32_rx_isr_Button pressed! Send message 1
handleButton pressed 23 times
r: Message on filter index 4
--- 5 messages dropped ---
[00:01:12.230,000] <dbg> stm32_can.can_stm32_send: Using mailbox 1
[00:01:12.230,000] <dbg> stm32_can.can_stm32_rx_isr_handler: Message on filter index 1
[00:01:12.230,000] <dbg> stm32_can.can_stm32_rx_isr_handler: Message on filter index 0
[00:01:12.250,000] <dbg> stm32_can.can_stm32_send: Sending 1 bytes on CAN_1. Id: 0x10, ID type: standard, Remote Frame: no
[00:01:12.250,000] <dbg> stm32_can.can_stm32_send: Using mailbox 0
[00:01:12.250,000] <dbg> stm32_can.can_stm32_send: Sending 2 bytes on CAN_1. Id: 0x1, ID type: standard, Remote Frame: no
[00:01:12.250,000] <dbg> stm32_can.can_stm32_send: Using mailbox 1
[00:01:12.250,000] <dbg> stm32_can.can_stm32_rx_isr_handler: Message on filter index 1
[00:01:12.2Button pressed! Send message 0
50,000Button pressed 24 times
]
........... and so on
```

alexanderwachter, 2019-04-13:
@SebastianStumm The logg looks good actually. Except it doesn't stop sending. I think this behavior is not related to the CAN driver. Can you run the test cases? The location is `zephyr/tests/drivers/can/api/`.
You can build and flash them with logging enabled. If you see SUCCESS at the end, everything is fine.

SebastianStumm, 2019-04-14:
@alexanderwachter I also don't think it's the CAN driver. I ran the tests with the following result:
```
▒[00:00:00.000,000] <dbg> stm32_can.can_stm32_runtime_configure: Runtime configure of CAN_1 done
--- 42 messages dropped ---
[00:00:00.000,000] <dbg> stm32_can.can_stm32_set_filter: Filter set! Filter number: 0 (index 0)
[00:00:00.000,000] <dbg> stm32_can.can_stm32_detach: Detatch filter number 0 (index 0), type 2
[00:00:00.000,000] <dbg> stm32_can.can_stm32_detach: Bank number 0 is empty -> deakivate
[00:00:00.000,000] <dbg> stm32_can.can_stm32_set_filter: Setting filter ID: 0x555, mask: 0x7f0
[00:00:00.000,000] <dbg> stm32_can.can_stm32_set_filter: Filter type: standard ID with mask (0)
[00:00:00.000,000] <dbg> stm32_can.can_stm32_set_filter: Filter set! Filter number: 0 (index 0)
[00:00:00.000,000] <dbg> stm32_can.can_stm32_detach: Detatch filter number 0 (index 0), type 0
[00:00:00.000,000] <dbg> stm32_can.can_stm32_detach: Bank number 0 is empty -> deakivate
[00:00:00.000,000] <dbg> stm32_can.can_stm32_set_filter: Setting filter ID: 0x15555555, mask: 0x1ffffff0
[00:00:00.000,000] <dbg> stm32_can.can_stm32_set_filter: Filter type: extended ID with mask (2)
[00:00:00.000,000] <dbg> stm32_can.can_stm32_set_filter: Filter set! Filter number: 0 (index 0)
--- 45 messages dropped ---
[00:00:00.110,000] <dbg> stm32_can.can_stm32_send: Using mailbox 0
--- 29 messages dropped ---
[00:00:00.120,000] <dbg> stm32_can.can_stm32_rx_isr_handler: Message on filter index 0
[00:00:00.120,000] <dbg> stm32_can.can_stm32_detach: Detatch filter number 0 (index 0), type 2
[00:00:00.120,000] <dbg> stm32_can.can_stm32_detach: Bank number 0 is empty -> deakivate
[00:00:00.120,000] <dbg> stm32_can.can_stm32_send: Sending 8 bytes on CAN_1. Id: 0x1555555a, ID type: extended, Remote Frame: no
[00:00:00.120,000] <dbg> stm32_can.can_stm32_send: Using mailbox 0
[00:00:00.120,000] <dbg> stm32_can.can_stm32_set_filter: Setting filter ID: 0x15555555, mask: 0x1ffffff0
[00:00:00.120,000] <dbg> stm32_can.can_stm32_set_filter: Filter type: extended ID with mask (2)
[00:00:00.120,000] <dbg> stm32_can.can_stm32_set_filter: Filter set! Filter number: 0 (index 0)
[00:00:00.120,000] <dbg> stm32_can.can_stm32_rx_isr_handler: Message on filter index 0
[00:00:00.120,000] <dbg> stm32_can.can_stm32_detach: Detatch filter number 0 (index 0), type 2
[00:00:00.120,000] <dbg> stm32_can.can_stm32_detach: Bank number 0 is empty -> deakivate
--- 1 messages dropped ---
===================================================================
starting test - test_send_receive_wrong_id
[00:00:00.120,000] <dbg> stm32_can.can_stm32_set_filter: Setting filter ID: 0x555, mask: 0x7ff
[00:00:00.120,000] <dbg> stm32_can.can_stm32_set_filter: Filter type: standard ID without mask (1)
[00:00:00.120,000] <dbg> stm32_can.can_stm32_set_filter: Filter set! Filter number: 0 (index 0)
[00:00:00.120,000] <dbg> stm32_can.can_stm32_send: Sending 8 bytes on CAN_1. Id: 0x55a, ID type: standard, Remote Frame: no
[00:00:00.120,000] <dbg> stm32_can.can_stm32_send: Using mailbox 0
[00:00:00.230,000] <dbg> stm32_can.can_stm32_detach: Detatch filter number 0 (index 0), type 1
[00:00:00.230,000] <dbg> stm32_can.can_stm32_detach: Bank number 0 is empty -> deakivate
PASS - test_send_receive_wrong_id
===================================================================
Test suite can_driver succeeded
===================================================================
**PROJECT EXECUTION SUCCESSFUL**
```

I also modified the sample by adding a work thread which sends "Hallo Welt" each 5 seconds and this seems to work just fine. So I assume the problem lays outside of the can driver itself, maybe the gpio/user-button.
```
▒***** Booting Zephyr OS v1.14.0-rc3-128-ged139948a54b *****
Finished init. waiting for Interrupts
TX thread is running.
filter id: 1
[00:00:00.000,000] <dbg> stm32_can.can_stm32_runtime_configure: Runtime configure of CAN_1 done
[00:00:00.000,000] <dbg> stm32_can.config_can_1_irq: Enable CAN1 IRQ
[00:00:00.000,000] <inf> stm32_can: Init of CAN_1 done
[00:00:00.000,000] <dbg> stm32_can.can_stm32_runtime_configure: Runtime configure of CAN_1 done
[00:00:00.000,000] <dbg> stm32_can.can_stm32_set_filter: Setting filter ID: 0x1, mask: 0x7ff
[00:00:00.000,000] <dbg> stm32_can.can_stm32_set_filter: Filter type: standard ID without mask (1)
[00:00:00.000,000] <dbg> stm32_can.can_stm32_set_filter: Filter set! Filter number: 0 (index 0)
[00:00:00.010,000] <dbg> stm32_can.can_stm32_set_filter: Setting filter ID: 0x10, mask: 0x7ff
[00:00:00.010,000] <dbg> stm32_can.can_stm32_set_filter: Filter type: standard ID without mask (1)
[00:00:00.010,000] <dbg> stm32_can.can_stm32_set_filter: Filter set! Filter number: 1 (index 1)
[00:00:00.010,000] <dbg> stm32_can.can_stm32_set_filter: Setting filter ID: 0x12345, mask: 0x1fffffff
[00:00:00.010,000] <dbg> stm32_can.can_stm32_set_filter: Filter type: extended ID without mask (3)
[00:00:00.010,000] <dbg> stm32_can.can_stm32_set_filter: Filter set! Filter number: 4 (index 4)
Sending ...
Hallo Welt
[00:00:05.000,000] <dbg> stm32_can.can_stm32_send: Sending 8 bytes on CAN_1. Id: 0x12345, ID type: extended, Remote Frame: no
[00:00:05.000,000] <dbg> stm32_can.can_stm32_send: Using mailbox 0
[00:00:05.000,000] <dbg> stm32_can.can_stm32_send: Sending 3 bytes on CAN_1. Id: 0x12345, ID type: extended, Remote Frame: no
[... 12 lines cut ...]
Hallo Welt
[... cut ...]
[00:00:25.000,000] <dbg> stm32_can.can_stm32_send: Using mailbox 0
[00:00:25.000,000] <dbg> stm32_can.can_stm32_send: Sending 3 bytes on CAN_1. Id: 0x12345, ID type: extended, Remote Frame: no
[00:00:25.000,000] <dbg> stm32_can.can_stm32_send: Using mailbox 1
[00:00:25.000,000] <dbg> stm32_can.can_stm32_rx_isr_handler: Message on filter index 4
[00:00:25.000,000] <dbg> stm32_can.can_stm32_rx_isr_handler: Message on filter index 4
```
Should I do a pr now for the CAN controller or try to figure out why it's dropping messages when using the user button first?

alexanderwachter, 2019-04-14:
>Should I do a pr now for the CAN controller or try to figure out why it's dropping messages when using the user button first?

Go for it!
If you have time to figure it out you can do that but as long as the CAN driver works, I'm fine with it.

alexanderwachter, 2019-05-09:
@lucaspeixotot Did you use CAN1 and CAN2 concurrently?

alexanderwachter, 2019-05-09:
@SebastianStumm @lucaspeixotot can you guys review #13262? Also please leave some comments about feature requests and general thoughts about the API.

alexanderwachter, 2019-07-19:
Think this has been settled.

closed 2019-07-19 by alexanderwachter