zephyrproject-rtos/zephyr #14942: tests/posix/fs don't build on em_starterkit_em11d

galak, 2019-03-27:
`./scripts/sanitycheck -T tests/posix/fs -p em_starterkit_em11d`

Gets:

```
JOBS: 64
Cleaning output directory /home/galak/git/zephyr/sanity-out
Building testcase defconfigs...
2 tests selected, 320 tests discarded due to filters
total complete:    0/   2   0%  failed:    0

em_starterkit_em11d       tests/posix/fs/portability.posix.newlib            FAILED: build_error
	see: sanity-out/em_starterkit_em11d/tests/posix/fs/portability.posix.newlib/build.log

total complete:    1/   2  50%  failed:    1

em_starterkit_em11d       tests/posix/fs/portability.posix                   FAILED: build_error
	see: sanity-out/em_starterkit_em11d/tests/posix/fs/portability.posix/build.log

total complete:    2/   2  100%  failed:    2
0 of 2 tests passed with 0 warnings in 3 seconds
```

galak, 2019-03-27:
We can easily exclude this board, but it seems like the configuration with its build about what we think is "FLASH" and what we think is "SRAM" might not be right.

```
Memory region         Used Size  Region Size  %age Used
            ICCM:          0 GB        64 KB      0.00%
            SRAM:      141532 B       128 KB    107.98%
            DCCM:          0 GB        64 KB      0.00%
        IDT_LIST:          57 B         2 KB      2.78/opt/zephyr-sdk-0.10.0/arc-zephyr-elf/bin/../lib/gcc/arc-zephyr-elf/8.3.0/../../../../arc-zephyr-elf/bin/ld: zephyr_prebuilt.elf section `noinit' will not fit in region `SRAM'
/opt/zephyr-sdk-0.10.0/arc-zephyr-elf/bin/../lib/gcc/arc-zephyr-elf/8.3.0/../../../../arc-zephyr-elf/bin/ld: region `SRAM' overflowed by 10460 bytes
collect2: error: ld returned 1 exit status
```

galak, 2019-03-27:
I'm not clear which memory on the em11d we expect to be running out of.  The dts seems to imply DDR.  `zephyr,sram = &ddr0;` But I think the linker scripts have a different idea.

ruuddw, 2019-03-28:
This is related to PR #14811.
Somewhere a /1024 is applied in the dts magic, and emsk linker script does additional /1024, resulting in 128k instead of 128M memory size.

Generated dts section:
`/* memory@10000000 */
#define DT_MMIO_SRAM_0				1
#define DT_MMIO_SRAM_10000000_BASE_ADDRESS	0x10000000
#define DT_MMIO_SRAM_10000000_SIZE		134217728
#define DT_SRAM_BASE_ADDRESS			0x10000000
#define DT_SRAM_SIZE				131072
#define DT_MMIO_SRAM_0_BASE_ADDRESS		DT_SRAM_BASE_ADDRESS
#define DT_MMIO_SRAM_0_SIZE			DT_SRAM_SIZE`

Any idea why the DT_SRAM_SIZE is 131072 instead of 134217728?
Something to fix in dts, or in the emsk linker script?

galak, 2019-03-28:
The linker script needs fixing.  We've always scaled DT_SRAM_SIZE by 1024 in Zephyr.  I want to change that, but haven't gotten around to cleaning it up.

PR #15010 (2019-03-29): board: emsk: fix the ddr related definitons
* add ddr fixup in dts_fixup.h
* use DT_DDR_XXX in linker.ld

Fixes #14942
closed 2019-03-29 by nashif