zephyrproject-rtos/zephyr #14943: config BOARD_HAS_NRF5_BOOTLOADER not honored for nrf52840_pca10059

markrages, 2019-03-27:
To make executables that load with the n5 bootloader, we have this config option in `boards/arm/nrf52840_pca10059/Kconfig`:

```
config BOARD_HAS_NRF5_BOOTLOADER
        bool "Board has nRF5 bootloader"
        default y
        help
         If selected, applications are linked so that they can be loaded by
         Nordic nRF5 bootloader.
```

But it doesn't work.  The executable is still linked at 0x0, not 0x1000.

**To Reproduce**
1. `west build -d build/nRF52/blinky -s zephyrproject/zephyr/samples/basic/blinky -b nrf52840_pca10059`
2. `head build/nRF52/blinky/zephyr/linker.cmd` gives this output:

```OUTPUT_FORMAT("elf32-littlearm")
_region_min_align = 32;
MEMORY
    {
    FLASH (rx) : ORIGIN = (0x0 + 0x0), LENGTH = (1024*1K - 0x0)
    SRAM (wx) : ORIGIN = 0x20000000, LENGTH = (256 * 1K)
    IDT_LIST (wx) : ORIGIN = (0x20000000 + (256 * 1K)), LENGTH = 2K
    }
ENTRY("__start")
SECTIONS
```

**Expected behavior**
FLASH (rx) : ORIGIN should start at 0x1000.

**Impact**
Cannot use zephyr executables with nRF-Connect or nrfutil.

**Environment**
 - OS: Ubuntu 18.04
 - Toolchain: zephyr-sdk-0.10.0
 - Commit: c3cf543452ca57c37e5ffb2a06731f23ed9d297e

markrages, 2019-03-27:
zephyr/.config has `CONFIG_TEXT_SECTION_OFFSET=0x1000` but this isn't actually applied when the linker script is generated.

The linker script uses `CONFIG_FLASH_BASE_ADDRESS=0x0` and ignores `TEXT_SECTION_OFFSET`.

lemrey, 2019-03-28:
Hi @markrages, could you please check if #14978 fixes this issue. Thank you.

markrages, 2019-03-28:
Fixed by #14978, thank you.

closed 2019-03-28 by carlescufi