zephyrproject-rtos/zephyr #14945: nrf52840_pca10059 executables do not work without mcuboot

markrages, 2019-03-27:
** Description **

The LED blink example build for nrf52840_pca10059 does not work when loaded with nRF Connect.

**To Reproduce**
Steps to reproduce the behavior:
1. `west build -d build/nRF52/blinky -s zephyrproject/zephyr/samples/basic/blinky -b nrf52840_pca10059 -- -DCONFIG_FLASH_BASE_ADDRESS=0x1000` (the define is necessitated by #14943)
2. Load executable into the dongle board with nRF Connect app.
3. The LED is not blinking on the target board.

**Expected behavior**

The LED blinks.

**Impact**
Prevents simple use of nRF-Connect app, or distribution of images.

** Workaround **
The LED can blinks if I use this workaround, suggested by @mike-scott on IRC.  
1. `west build -d build/nRF52/mcuboot -s mcuboot/boot/zephyr -b nrf52840_pca10059 -- -DCONFIG_FLASH_LOAD_OFFSET=0x1000 -DCONFIG_FLASH_LOAD_SIZE=0xf000`
2. `west build -d build/nRF52/blinky -s zephyrproject/zephyr/samples/basic/blinky -b nrf52840_pca10059 -- -DCONFIG_FLASH_LOAD_OFFSET=0x10000 -DCONFIG_FLASH_LOAD_SIZE=0x5E000`
3. `west sign  -d build/nRF52/blinky -t imgtool -- --key mcuboot/root-rsa-2048.pem`
4. load `build/nRF52/mcuboot/zephyr/zephyr.hex` and `zephyr.signed.hex` simultaneously with nRF Connect app.

**Environment**
 - OS: Ubuntu 18.04
 - Toolchain: zephyr-sdk-0.10.0
 - Commit: c3cf543452ca57c37e5ffb2a06731f23ed9d297e

lemrey, 2019-03-28:
This issue does not seem valid, since the "nRF Connect App" is not documented as a supported DFU method for this board. The support DFU method is via `nrfutil`.

markrages, 2019-03-28:
@lemrey edited to show nrfutil use.  The file is linked incorrectly, so the programmer script makes no difference.

lemrey, 2019-03-28:
This is issue is still confusing, since you apply a workaround and argue that the result is not what expected; you are not supposed to manually override the `FLASH_LOAD_OFFSET`.

I think all the issues you have opened are rooted in one, real issue: https://github.com/zephyrproject-rtos/zephyr/issues/14943,
for which I have opened a PR: https://github.com/zephyrproject-rtos/zephyr/pull/14978.

Please check if that fixes it.

markrages, 2019-03-28:
@lemrey I agree, my three issues all stem from #14943 (and #14875.)

I will close them all when #14978 is merged.

carlescufi, 2019-03-28:
Closing, #14978 is merged

closed 2019-03-28 by carlescufi