zephyrproject-rtos/zephyr #14946: cdc_acm example doesn't work on nrf52840_pca10059

markrages, 2019-03-27:
**Describe the bug**
CDC_ACM example doesn't work in nrf52840_pca10059.  The USB device doesn't appear.

**To Reproduce**
I build using the workaround in #14945:
1. `west build -d build/nRF52/mcuboot -s mcuboot/boot/zephyr -b nrf52840_pca10059 -- -DCONFIG_FLASH_LOAD_OFFSET=0x1000 -DCONFIG_FLASH_LOAD_SIZE=0xf000`
2. `west build -d build/nRF52/cdc_acm -s zephyrproject/zephyr/samples/subsys/usb/cdc_acm -b nrf52840_pca10059 -- -DCONFIG_FLASH_LOAD_OFFSET=0x10000 -DCONFIG_FLASH_LOAD_SIZE=0x5E000`
3. `west sign  -d build/nRF52/cdc_acm/ -t imgtool -- --key mcuboot/root-rsa-2048.pem`
4. Load `build/nRF52/mcuboot/zephyr/zephyr.hex` and `zephyr.signed.hex` simultaneously with nRF Connect app.

**Expected behavior**
A device appears in `lsusb` output.  A tty device appears as `/dev/ttyACM*`

**Actual behavior**
No device appears in `lsusb`.  No USB messages appear in `dmesg`.

**Impact**
I am stuck in my development without a working serial port.

**Environment**
 - OS: Ubuntu 18.04
 - Toolchain: zephyr-sdk-0.10.0
 - Commit: c3cf543452ca57c37e5ffb2a06731f23ed9d297e

markrages, 2019-03-27:
Here's backtrace:

```(gdb) bt
#0  0x00017edc in cdc_acm_line_ctrl_get (dev=0x20003764 <__device_cdc_acm_0>, 
    ctrl=4, val=0x2000239c <_main_stack+980>)
    at /home/markrages/zephyr/zephyrproject/zephyr/subsys/usb/class/cdc_acm.c:878
#1  0x0001154c in z_impl_uart_line_ctrl_get (val=0x2000239c <_main_stack+980>, 
    ctrl=4, dev=0x20003764 <__device_cdc_acm_0>)
    at /home/markrages/zephyr/zephyrproject/zephyr/include/uart.h:1112
#2  uart_line_ctrl_get (ctrl=4, val=0x2000239c <_main_stack+980>, 
    dev=0x20003764 <__device_cdc_acm_0>)
    at zephyr/include/generated/syscalls/uart.h:41
#3  main ()
    at /home/markrages/zephyr/zephyrproject/zephyr/samples/subsys/usb/cdc_acm/src/main.c:89
```

markrages, 2019-03-28:
Waiting for DTR is like #13662. Except I am sure the device is plugged in to USB -- that's how I loaded the code.

jfischer-phytec-iot, 2019-03-28:
>Except I am sure the device is plugged in to USB -- that's how I loaded the code.

micro USB connector labeled as "nRF USB" is also connected to the host?

markrages, 2019-03-28:
@jfischer-phytec-iot the '59 is the dongle, see picture on this unrelated bug: https://github.com/NordicSemiconductor/pc-nrfconnect-programmer/issues/116#issuecomment-477321052

markrages, 2019-03-28:
So, this worked once.

I did the `git bisect` dance, and learned two things:
1. usb is not working with the #1495 workaround. Instead compile with `west build -d $BUILDDIR -s $SOURCEDIR -b nrf52840_pca10059 -- -DCONFIG_FLASH_LOAD_OFFSET=0x1000 -DCONFIG_FLASH_LOAD_SIZE=0x1f000`
2. `cdc_acm` example works before 5977b3db2f29d169adcf032149be4cab94964829, and is broken from then on.

jfischer-phytec-iot, 2019-03-28:
Tested with reel board and frdm-k64f, works fine.

lemrey, 2019-03-28:
Hi @markrages, could you please check if https://github.com/zephyrproject-rtos/zephyr/pull/14978 fixes this issue. Thank you.

markrages, 2019-03-28:
@lemrey yes, #14978 fixes it.

markrages, 2019-03-28:
The #14945 workaround did not work for USB example, I assume because https://github.com/zephyrproject-rtos/zephyr/pull/14875

carlescufi, 2019-03-28:
@markrages is this one fixed too by #14875? If so, could you please close it? Thanks!

markrages, 2019-03-28:
Everything appears to work with #14978 and #14875 merged.

closed 2019-03-28 by markrages