zephyrproject-rtos/zephyr #14957: [Coverity CID :197458]Insecure data handling in /subsys/usb/usb_device.c

aasthagr, 2019-03-27:
Static code scan issues seen in File: /subsys/usb/usb_device.c
 Category: Insecure data handling
 Function: usb_set_interface
 Component: Other
 CID: 197458
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

jfischer-phytec-iot, 2019-03-27:
I think we need appropriate macros in usb_device.h, something like:

```
#define USB_MPS_MAX_FS_BULD 64
#define USB_MPS_MAX_HS_BULD 512
...
#define USB_MPS_MAX_FS_INT 64
```

and checks for that in `bool usb_set_interface(u8_t iface, u8_t alt_setting)` or `usb_dc_ep_configure(const struct usb_dc_ep_cfg_data * const cfg)`.

finikorg, 2019-03-28:
So we continue to check endpoint address at device controllers and MPS in usb_device, right?

finikorg, 2019-03-28:
Or maybe we can have 
```
#if FS
#define USB_MAX_BULK_MPS 64
#else
...
```

jfischer-phytec-iot, 2019-03-28:
I suggest to add a common usb_dc_validate_ep_config(struct usb_dc_ep_cfg_data *ep_cfg, bool speed) function and call it from the drivers. 

```c
int usb_dc_validate_ep_config(struct usb_dc_ep_cfg_data *ep_cfg, bool high_speed)
{
  if (high_speed)  {
       /* check for the endpoint type and mps */
       .....
        if (ep_cfg->ep_mps > USB_MPS_MAX_HS_BULK) {
             return -1;
        }
       ......
  }
  return 0;
}
```

finikorg, 2019-03-28:
@jfischer-phytec-iot `usb_dc_validate_ep_config()` would make impression that this is device controller function. It would be called only from set_endpoint(), or do we want it to be called from device controllers?

commit d2c6869e (2019-04-11): drivers: usb: check maximum packet size

Check maximum packet size (MPS) of an endpoint in usb_dc_kinetis
and usb_dc_native_posix drivers.

resolves #14957
resolves #14954
closed 2019-04-11 by nashif