zephyrproject-rtos/zephyr #14970: samples/drivers/watchdog: Not showing "Waiting for reset..." for nrf boards

cinlyooi-intel, 2019-03-28:
**Describe the bug**
We are expecting to see the line "Waiting for reset...". However, we did not see it.

**To Reproduce**
Affected boards are 
 - nrf51_pca10028 (frequency: occasional) 
 - nrf52_pca10040 (frequency: 50:50)
 - nrf52840_pca10056 (frequency: almost always affected)

Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=nrf52840_pca10056
3. make
4. make run

**Screenshots or console output**
```
Feeding watchdog 5 times
Feeding watchdog...
Feeding watchdog...
Feeding watchdog...
Feeding watchdog...
Feeding watchdog...
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-1514-gcf91f8cf9001 (delayed boot 1000ms) *****
Watchdog sample application
Feeding watchdog 5 times
Feeding watchdog...
Feeding watchdog...
Feeding watchdog...
Feeding watchdog...
Feeding watchdog...
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-1514-gcf91f8cf9001 (delayed boot 1000ms) *****
Watchdog sample application
Feeding watchdog 5 times
Feeding watchdog...
Feeding watchdog...
```

I am expecting `Waiting for reset..` before the second and subsequent `***delay boot` banners.

**Environment (please complete the following information):**
 - OS: fedora29
 - Toolchain: zephyr
 - Commit SHA or Version used: cf91f8cf9001bb31b3c8770104a516b1567a8636

**Additional context**
- We see reset being performed, i.e., multiple delay boot/boot banners.
    - This is expected and correct behaviour, since we want watchdog to reboot after 5000ms of inactivity
- Sometimes we do see the "Waiting for reset...", but it is very rare, 1 in 50 perhaps.
- Experimentation says the MCU is resetted inside the last k_sleep(1000) call.

carlescufi, 2019-03-29:
@kl-cruz could you please take a look at this?

kl-cruz, 2019-04-02:
@carlescufi  I'll have some time 8th April. If it is important please try to ask someone else

overheat, 2019-04-12:
@cinlyooi-intel I works with today's source code version, I suppose this bug be fixed by other commits. FYI.
[image]

ioannisg, 2019-06-30:
This is fixed now. as @overheat suggested. I tested on all nRF SoCs.

closed 2019-06-30 by ioannisg