zephyrproject-rtos/zephyr #14988: USB device not recognized in USB DFU sample

Qbicz, 2019-03-28:
**Describe the bug**
nrf52840_pca10056, USB DFU sample. USB device used to be recognized in the past, with recent Zephyr it's unusable with following message on Windows:

`USB Device Not Recognized. One of the USB devices attached to this computer has malfunctioned and Windows does not recognize it. For assistance in solving this problem, click this message.`

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Erase the board
2. Flash MCUBoot built for DBOARD=nrf52840_pca10056
2. Compile USB DFU sample
3. Sign sample with imgtool
4. Flash signed sample next to bootloader
5. Attach USB cable

**Environment (please complete the following information):**
 - built on Ubuntu 16.04 LTS
 - Toolchain (e.g Zephyr SDK 0.10.0)
 - Commit SHA 373a42b861
 - nrf52840_pca10056

Qbicz, 2019-03-28:
@nvlsianpu FYI

Qbicz, 2019-03-28:
I tried going back in history with bisect, but all old revisions resulted in the same problem. Toolchain issue? Tried on a few different kits.

carlescufi, 2019-03-28:
This sounds like it needs an `.inf` file. @masz-nordic?

masz-nordic, 2019-03-28:
No, this message appears when the device fails during enumeration.
I was actually unable to reproduce with Zephyr 373a42b and MCUBoot ebd0503 - device is enumerated properly.

carlescufi, 2019-03-28:
Setting prio as low since this impacts a single sample, not a single board.

Qbicz, 2019-03-29:
This issue occurs on PCA10056 preview DKs, version 0.9.0 and 0.10.0. These are engineering samples which are obsolete anyway.

On PCA10056 0.12.0 and 1.0.0 problem cannot be reproduced.

carlescufi, 2019-03-29:
Closing since this is a hardware problem

closed 2019-03-29 by carlescufi