zephyrproject-rtos/zephyr #14997: Convert samples/sensor/bme280/src/main.c to use printk instead of printf

galak, 2019-03-28:
Replace printf with printk in samples/sensor/bme280/src/main.c

balajikulkarni, 2019-04-10:
I have been following Zephyr and going through it's internals for sometime now, and I wish to start contributing to Zephyr. Could you please help me on taking up my first task?

Thea-Aldrich, 2019-04-10:
Hello and welcome to the Zephyr Project! It is great to hear you are interested in contributing. This is a great first issue. Have you taken a look at the [Contribution Guidelines](https://docs.zephyrproject.org/latest/contribute/index.html)? You'll be able to find the coding style and commit guidelines there, as well as information about our CI, contributions tools and workflow. I'm happy to help along the way. We also have a great Slack channel and developer mailing list (links below) where you can chat with other contributors and ask questions.
Best,
Thea

Slack: https://join.slack.com/t/zephyrproject/shared_invite/enQtNDgwNzk5MTU3NTA3LWNhZjc4MzM4Nzg5NGRiNTA2OGVhN2ZlNzZmYWE5Yjc2ZDIwYjYzNGJmMzNiYmY2MTdiYmQ1M2I4NGEzMmUzZjA

Public Meeting: https://www.zephyrproject.org/developers/meetings/

Developer Mailing List: devel@lists.zephyrproject.org
--

balajikulkarni, 2019-04-11:
Thank you. I think this issue is already fixed. Need to look at other good first issues. How do show interest in picking up issues for which I could provide a fix? so that no one else is working on the same.

Thea-Aldrich, 2019-04-11:
We have a couple ways that you can see who is working on an issue and get more information about other solutions that may be being considered. Often times, when a solution to an issue is being actively worked on, you'll see reference to a PR that attempts to solve the issue. In this case: https://github.com/zephyrproject-rtos/zephyr/pull/14998. 

A good place to start asking is to contact the issue owner to see if they have any insight into current work on the issue. We also have a lot of discussions on Slack and the developer mailing list where new contributors can ask around and also talk about their proposed solution to the issue. This is especially helpful when an issue has been sitting for a while or is part of a larger change that may be coming down the pipeline. 

The Thursday dev review meeting is another great place to get synced around issues and PR's. Most of the core developers and maintainers attend this call and go over new issues and PR's and coordinate work across organizations. The team is very friendly and welcoming. Its a smaller group than the TSC call.

PR #14998 (2019-04-18): samples: sensor: bme280: Convert code to use printk
There isn't any specific need to use printf here, convert to use printk
instead.

Fixes #14997
commit 6b4007e5 (2019-04-18): samples: sensor: bme280: Convert code to use printk

There isn't any specific need to use printf here, convert to use printk
instead.

Fixes #14997
closed 2019-04-18 by galak