zephyrproject-rtos/zephyr #15003: init_iface in net_if.c should check api->init

vonhust, 2019-03-29:
**Describe the bug**
the init_ifcae should check api->init is not NULL before calling it. 

Although `NET_ASSERT(api && api->init)` is added to check, but if ASSERT is not enable and api->init is null,  this will cause sw restart.

A possible case of `api->init == null` is, if the hardware init of net interface called in `z_sys_device_do_config_level` failed, i.e retval !=0, then `info->driver_api` will be set null.  

A fix is
`
if (api->init) {
  api->init(iface);
}
`

jukkar, 2019-03-29:
If the device configuration fails, then obviously the interface cannot be used. I just wonder how the init pointer can be null as typically it is set at compile time by `NET_DEVICE_INIT()` and should be always available. Could you point to the sources of the device driver that causes this issue?

vonhust, 2019-04-01:
@jukkar
the device related info may be cleared in kernel/device.c line 61
```
		retval = device_conf->init(info);
		if (retval != 0) {
			/* Initialization failed. Clear the API struct so that
			 * device_get_binding() will not succeed for it.
			 */
line61:	 info->driver_api = NULL;
```
the failure will not stop the call of `net_init->init_rx_queue->net_if_init`, then

const struct net_if_api *api = net_if_get_device(iface)->driver_api = NULL (set by line61);

jukkar, 2019-04-01:
Indeed, that is unfortunate that the API struct is cleared. Can you propose a patch that adds the check to net_if.c?

PR #15471 (2019-05-07): net: Check device driver API pointer
It is possible that the device driver API pointer is null.
For example if the device driver returns an error, the device
code will make the API pointer NULL so that the API would not
be used. This can cause errors in networking code where we
typically do not check the NULL value.

Fixes #15003
commit 9a6bbbfb (2019-05-07): net: Check device driver API pointer

It is possible that the device driver API pointer is null.
For example if the device driver returns an error, the device
code will make the API pointer NULL so that the API would not
be used. This can cause errors in networking code where we
typically do not check the NULL value.

Fixes #15003
closed 2019-05-07 by jukkar