zephyrproject-rtos/zephyr #15009: sanitycheck --coverage on qemu_x86:  cannot move location counter backwards

cinlyooi-intel, 2019-03-29:
*** Description ***
When trying to run coverage check, linker failed with `cannot move location counter backwards (from 0000000000047b1c to 0000000000047774)`

*** To Reproduce ***
```
$ ./scripts/sanitycheck --coverage --coverage-platform qemu_x86 -p qemu_x86 --gcov-tool /opt/zephyr-sdk-0.10.0/i586-zephyr-elf/bin/i586-zephyr-elf-gcov -T tests/kernel/common  | tee output.log
JOBS: 8
Cleaning output directory /home/cinlyooi/tests/zephyr.git/sanity-out
Building testcase defconfigs...
2 tests selected, 322 tests discarded due to filters
1/2 qemu_x86                  tests/kernel/common/kernel.common                  FAILED: handler_crash
	see: sanity-out/qemu_x86/tests/kernel/common/kernel.common/build.log
2/2 qemu_x86                  tests/kernel/common/kernel.common.misra            FAILED: handler_crash
	see: sanity-out/qemu_x86/tests/kernel/common/kernel.common.misra/build.log
Generating coverage files...
geninfo: WARNING: no .gcda files found in /home/cinlyooi/tests/zephyr.git/sanity-out - skipping!
lcov: ERROR: no valid records found in tracefile /home/cinlyooi/tests/zephyr.git/sanity-out/coverage.info
lcov: ERROR: no valid records found in tracefile /home/cinlyooi/tests/zephyr.git/sanity-out/coverage.info
lcov: ERROR: no valid records found in tracefile /home/cinlyooi/tests/zephyr.git/sanity-out/coverage.info
lcov: ERROR: no valid records found in tracefile /home/cinlyooi/tests/zephyr.git/sanity-out/coverage.info
genhtml: ERROR: no valid records found in tracefile /home/cinlyooi/tests/zephyr.git/sanity-out/coverage.info
0 of 2 tests passed with 0 warnings in 8 seconds

$ cat sanity-out/qemu_x86/tests/kernel/common/kernel.common/build.log
# <snip>
[ 98%] Building C object zephyr/CMakeFiles/kernel_elf.dir/misc/empty_file.c.obj
[100%] Linking C executable zephyr.elf
/opt/zephyr-sdk-0.10.0/i586-zephyr-elf/bin/../lib/gcc/i586-zephyr-elf/8.3.0/../../../../i586-zephyr-elf/bin/ld:linker_pass_final.cm
d:65 cannot move location counter backwards (from 0000000000047b1c to 0000000000047774)
collect2: error: ld returned 1 exit status
make[3]: *** [zephyr/CMakeFiles/kernel_elf.dir/build.make:102: zephyr/zephyr.elf] Error 1

#<snip>
```
*** Info ***
OS: fedora29
ToolChain: zephyr
CommitID: f5a4a698ec

andrewboie, 2019-03-29:
The fix for this is to increase CONFIG_KOBJECT_TEXT_AREA.

wentongwu, 2019-04-01:
PR #15005 should fix this issue.

commit 84b1bba8 (2019-04-02): codecov: keep "inline" for ALWAYS_INLINE

Previous commit c31e659165e9b55a5712f9298a63782de1725878 changed
the ALWAYS_INLINE macros to avoid functions being inlined for
the purpose of code coverage. This has a side effect of causing
the text sections of kobject_hash.c and priv_stacks_hash.c
to ballon more than 10 times in size. This is caused by
attaching the unused attribute, which results in all those
functions being in the text sections though they are never
used. So just keep the "inline" there.

This also removes -fno-inline from NO_COVERAGE_FLAGS so these
two files are not compiled with  [... cut]
closed 2019-04-02 by nashif